MyRobotLab/myrobotlab #395: autodisable function not responding

hairygael, 2018-12-12:
**Describe the bug**
Seems when there is crash with python, it removes the autodisable function for further gestures on servo.

**To Reproduce**
Steps to reproduce the behavior:
1. Start INMOOV_START
2. Run the gesture "Giving", which will throw an error related to a python piece being not defined.
3. Run "Rest" gesture
4. See the servos being not disabled after the proper timing.

**Expected behavior**
Even with python error, if you execute another gesture which doesn't throw an error, the servo should then be disabled.

**Desktop (please complete the following information):**
 - OS: windows 10 pro
 - Browser:chrome, safari
 - Version 1.1.56

moz4r, 2018-12-13:
Oh yes. When a gesture start, startedGesture() is called.
At the end finishedGesture() , only if there is no crash between...
Welcome ( soon ) to i01.execGesture() that will take care off crash and report it without break things.

hairygael, 2018-12-14:
Thanks for responding!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2018-12-18:
That is a sample to change a little how we can launch gestures, to talk about : 

https://github.com/MyRobotLab/inmoov/commit/33ff862acbf50d97f2bc20d0c3b50c751ec18345

You will see giving will not contain anymore started/stopped gesture, 2 advantages less code lines, and gestures chain without warning ( giving can call fullspeed by exemple )
The whole chain will be considered as 1 gesture.

execGesture will call for startedGesture() at the begining ( today is hardcoded and disable "autoDisable", we can enhance this )
and stopedGesture() at the end that end stuffs.
If there is an error in the py code, you will be vocally warned about, and also logged for more details. But the gesture will "finish" now.

We can call `i01.execGesture("giving()")` from script
or, `<oob><mrl><service>i01</service><method>execGesture</method><param>giving()</param></mrl></oob>`

hairygael, 2018-12-19:
> MyRobotLab/inmoov@33ff862

Hello Anthony,
That seems fine to me. Lots of AIML to check and to adapt. :)
Maybe the warning for the python error could be an option?
When working on coding, we maybe don't want to get recalled about the
warning every now and then.
As a side note: Just tested version 1.1.66 and got a python error which
froze the cmd.exe, but MRL was still responding.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2018-12-20:
You right ! vocal warning replaced by a "classical" one.