MyRobotLab/myrobotlab #398: OpenCV no longer starts with startHead in InMoov service.

kwatters, 2018-12-15:
For some reason , the InMoov service doesn't start OpenCV any more.  I discovered this when I tried called   i01.startAll(leftPort,rightPort) 

Normally it would be started if you start the InMoov Head service.

kwatters, 2018-12-15:
the startOpenCV method on the inmoov service also is calling a load and start call.  it should not.  
If you are starting an InMoov service you should programmatically configure your services..  or explicitly call runtime load and start from the python node.  it should *NOT* be hard coded in the JAVA code.

kwatters, 2018-12-15:
Additionally, why are we calling vision.test()    
Logic like this gets in the way if you want to have custom configurations.

moz4r, 2018-12-15:
Hi Kwatters ! I have some mods not yet pushed
About load() this is very important if we want to use UI ( swing or later to prepare for a webgui ) to configure services. It will load by exemple last user language, or camera index.
About vision.test() : it test for a worky opencv, and robot will speak an error to user if it fail.

kwatters, 2018-12-15:
As a user, I should never be forced to call load() on a service if I only want to start it.  If there is any code that forces you to call load() it needs to be removed.  If a user wants to call load() on a service, they should do it from python by calling load() ...  

As for the vision.test() method.. i think the intention of this method is good, however the implementation stops capture .. that is a side effect of this method.   Why not just rely on the "isCapturing()" method ?  Seems overly complex to have this additional logic.  (especially considering the side effect of calling test() that stops the camera on you.. that just seems wrong to me.)

kwatters, 2018-12-15:
This is verified to be worky again for harry&lloyd.

closed 2018-12-15 by kwatters

moz4r, 2018-12-15:
will add new methods like i01.loadAndStartAll() , also will not cameraOff() at startup, good for you ?

kwatters, 2018-12-15:
Why bother?  Just call i01.load() ... Then call i01.startX. ...

Less code is better...  Keep it clean. If you want to load a service... Do it in python