MyRobotLab/myrobotlab #403: RobotIsOpenCvCapturing missing when tracking in a gesture

hairygael, 2018-12-16:
**Describe the bug**
File "<string>", line 6, in relax
AttributeError: 'org.myrobotlab.service.InMoov' object has no attribute 'RobotIsOpenCvCapturing'

**To Reproduce**
Steps to reproduce the behavior:
1. START_INMOOV with latest inmoov-devellop and MRL version 1.1.61
2. Voice command RANDOM LIFE or RELAX
3. This returns an error related to RobotIsOpenCvCapturing that has no more attribute
4. 

**Expected behavior**
I think if RobotIsOpenCvCapturing is gone, we should have an alternative for gestures which can run along with tracking or opencv capturing. But I guess Anthony thought of that.
So this is mostly as a reminder.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: windows 10 pro
 - Browser chrome
 - Version 1.1.61

kwatters, 2018-12-16:
I saw a no worky here: http://myrobotlab.org/myrobotlab_log/upload/hairygael/1544996545.myrobotlab.log
this is related to some changes that moz4r and I have that caused this issue.  I have a updated the code to avoid this issue by having the startOpenCV() method return a boolean.   (True)

moz4r, 2018-12-16:
errors related to RobotIsOpenCvCapturing fixed . renamed to isCameraOn()

hairygael, 2018-12-16:
Gee!
Both faster to respond than an arrow!
Thanks

hairygael, 2018-12-16:
so we will have
if (i01.isCameraOn()):

kwatters, 2018-12-16:
Latest cookies are in the oven now..

hairygael, 2018-12-16:
Apparently not fixed in the inmoov repo
https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/gestures/relax.py

hairygael, 2018-12-16:
https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/gestures/giving.py

moz4r, 2018-12-16:
kwatters that is not enough I updated lot off gestures related to it. also I think this will conflict my last pending pr.