MyRobotLab/myrobotlab #419: savePredicates fails due to odd interactions with ProgramAB api.

kwatters, 2019-01-25:
There is a no worky from Gael that shows savePredicates can throw exceptions when switching sessions.

This is a nasty one because the various APIs are inconsistent and allow for scenarios where the path of the ProgramAB bot can end up being null.  There's also lots of other special logic that causes strange and unexepected behaviors when using ProgramAB.

This ticket will cover the code cleanup.

kwatters, 2019-01-26:
changes were merged in.  There were so many changes to the code base, I think this issue is probably solved now.  I'd like to close this ticket, but maybe @hairygael can test with the latest and see if it shows up again.

hairygael, 2019-01-30:
testing version 1.1.98
there is a issue with programAB not properly starting a session with random result.

To reproduce:
START_IMNOOV.bat
Once it is started, write in the chatbot "my name is John"
Sometimes the button "session" turns immediately green and all is fine, but sometimes the button staysorange and freezes MRL.
What is expected:
It should aways turn green and start properly the session and therefore create the predicate file in the directory config.

moz4r, 2019-01-31:
Before to dig, I confirm an issue ( not this one but related ) .

how is was worky before : 

 `<get name="name" />` Simply reflected the username session.  `<get name="name" />` is used everywhere for user identification. Like "what is my name"

Now : 

 `<get name="name" />` is "random"

- my name is johnDoo > `startSession("johnDoo") ` > what is my name ? >( `<get name="name" />`) johnDoo : **OK**
- my name is anthony> `startSession("anthony") ` > what is my name ? >( `<get name="name" />`) anthony : **OK**
- my name is johnDoo > `startSession("johnDoo ") ` > what is my name ? >( `<get name="name" />`) anthony : **NOK**

moz4r, 2019-02-08:
@kwatters , any output about inconsistant session names ? I'm not sure if it is a bug or the new logic you implemented about sessions.

supertick, 2021-12-20:
Is this still an issue ?

kwatters, 2021-12-21:
Closing this issue, our predicate saving has been completely redesigned.  If there are new issues, we can open a new ticket for it.

closed 2021-12-21 by kwatters