MyRobotLab/myrobotlab #427: Remove ServoEvents by Default - use explicit setting of encoder types

supertick, 2019-01-31:
Currently : Servo's expect to have ServoEvents.
ServoEvents are just Time-based Dead Recogning Pseudo Encoders.   This is ok, and very useful, but it should be treated as a more generalized Encoder.

e.g.  servoController.setEncoder(myEncoder) .. where myEncoder can be a "real" quadrature encoder or the TimeBased encoder;

I think initially Servos should have no encoders, and its the responsibility of the script or composite service to set the type of encoder. (The same exact rules should be followed for Motors & MotorControllers)

supertick, 2019-01-31:
Should be done with other massive refactoring for Servo & InMoov in Ogre

supertick, 2019-01-31:
When a Servo currently starts it automatically adds itself as a servo event listener and sets itself to publish ServoEvents.   I tried to create a static global default which configured all newly created Servos with preferences (e.g. without ServoEvents) but this was overwritten when the servo started :(

kwatters, 2019-01-31:
Did you want to close this or move it to Ogre?

supertick, 2019-01-31:
move to Ogre