MyRobotLab/myrobotlab #444: myrobotla.jar 1.1.121 won't install

hairygael, 2019-03-06:
It's been an issue since version 1.1.112, when double clicking myrobotlab.jar, it just doesn't do anything.
Thinking it was maybe related to my java 64bit version being obsolete, but installing java 64bit version 205, doesn't change the problem.
This is going on with two of my Windows 10 PC 64 bit.

kwatters, 2019-03-06:
I have reproduced the error here also.  
@supertick do you think this is related to any of your check-ins ?  I'm pretty sure when my resourceDir changes when in, that I validated a full download & install all ... and it was working. so I think this happened since then.

kwatters, 2019-03-07:
Thanks for the bug report!  Grog identified the fix.  the latest should be good again.

closed 2019-03-07 by kwatters