MyRobotLab/myrobotlab #450: Ear is borked

hairygael, 2019-03-08:
Version 1.1.126 when running START_INMOOV.bat
Ear is borked in Swingui and in Google Chrome webkitrecognition. (webgui)

hairygael, 2019-03-09:
tested version 1.1.127 it's a little better.
Chrome mic appears red for a moment but doesn't hear when speaking in the microphone.
Also i01.ear tab shows no user interface, when it should show some buttons and options.
[image]

hairygael, 2019-03-10:
tested version 1.1.129 it's better.
Chrome mic appears and speaking in the microphone works, words are recognized. Robot responds.
i01.ear tab shows no user interface, when it should show some buttons and options.

supertick, 2022-02-06:
fixed long ago

closed 2022-02-06 by supertick