MyRobotLab/myrobotlab #457: START_INMOOV.bat can't find myrobotlab.jar

hairygael, 2019-03-17:
Hello,
Testing version 1.1.133
When launching the START_INMOOV.bat file, the prompt throws the error: 
Error: Unable to access jarfile myrobotlab.jar
And then it closes the prompt command.

My tree directory is like this (since ages):
C:/Documents/myrobotlab/myrobotlab.1.1.133/myrobotlab.jar

supertick, 2019-03-17:
Maybe it was a bad download ?
I don't get this - when I install latest 133 in new directory .. do install ... START_INMOOV.bat gives me this ...

[image]

hairygael, 2019-03-17:
Ah mmhh...
Ok I will try to re download and re install and let you know.
:)

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

hairygael, 2019-03-17:
New download, new install, it's worky!
We can close this ticket.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2019-03-17 by supertick