MyRobotLab/myrobotlab #478: setAutoDisable seems borked

hairygael, 2019-04-02:
Testing version 1.1.150 with InMoov-develop
When launching the START_INMOOV.bat all servo are supposed to be setAutoDisable as True,
but it doesn't happen.
Looking on swingui the buttons of all servo are set to setAutoDisable : false
It appears you can't manually do it either in the swingui.

supertick, 2019-04-02:
Hi Gael,
As a warning servo.moveToBlocking is now (temporarily) bork'd too.   
Servo is getting new capabilities to handle any type of encoder (digital, analog, & kwatters' amt models)
You might want to hold off testing until we get this updated.
Thanks.

supertick, 2019-04-02:
Fixing these things now ....

hairygael, 2019-04-02:
Ok cool

supertick, 2022-02-06:
fixed long ago i believe

closed 2022-02-06 by supertick