MyRobotLab/myrobotlab #480: Audiofile problem to stream

hairygael, 2019-04-02:
Hello,
Testing version 1.1.150 with InMoov-develop
ERROR c.myrobotlab.audio.AudioProcessor - could not get audio input stream from input file
javax.sound.sampled.UnsupportedAudioFileException: could not get audio input stream from input file

I have sent a noworky

kwatters, 2019-04-02:
I saw the no worky.. it says the audio processor was having problems reading a file for playback.
Can you check if the file
C:\MRL\myrobotlab.1.1.150\audioFile\NaturalReaderSpeech\Matthew\83ab6be49bda6130bf0f95f6d58010ac.mp3 exist on your computer? and does it seem to be a valid mp3 ?

kwatters, 2019-04-03:
I reproduced this locally and it would seem that natural reader is returning an error message when trying to generate the speech.

{"rst":"ERR_CONVERT_FAILED"}

So, I think that for some reason the natural reader api is no longer available, or has changed so we'll need to update how we use it.

hairygael, 2019-04-03:
Argh..!
You are correct, the NaturalReader api must have been changed.
As a test, I removed all the audiofiles from a previous worky version, and
launched it, as a result: no audiofiles are working...
We are going to have tons of emails and messages from users that wonder why
their robots don't talk anymore.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2019-04-18:
I set the service as being "not available" - this is closed, as is yet another previously "free" voice resource endpoint (Google, Acapela now NaturalReader)

We might consider again a mrl key that would give access to paid systems - the cost of licensing paid subscriptions could be reduced through patreon or some other means of "group" support.

closed 2019-04-18 by supertick