MyRobotLab/myrobotlab #486: Servo / ServoGui attach fails

MatsMRL, 2019-04-07:
Attaching from the ServoGui no longer works. 
did not find method - servo.attach(ard,0,Double)
It should be possible to attach/detach using both the name or the ServoController
as it was in previous releases

supertick, 2019-04-07:
Thanks Mats, I'm working on this now .. 
I agree attach/detach should be able to use the a direct reference or a "name" - but that's not the problem.
The problem has to do with changes in the interface parameters to primitives.
Although, I thought I had put some implementation in to allow invoking based on primitives being coerced it failed.

I think kwatters was interested in "tightening" down the interface to primitives to avoid nulls.  But this does not work, and even when it did, its less efficient.  If you look into the details of invoking you find that it throws up, then tries to up-cast ... I did the upcast part, but not the coercion to primitives.

I'll write a post on MRL describing how interfaces which define service functions should not use primitives, but always the boxed counterparts e.g. Integer vs int - or somebody do a proficient method invoking/caching based on coercion. 

I'll be fixing this by making sure all service interfaces do not have primitives

supertick, 2019-04-07:
Gah ! ... my mistake 
glad I read your issue - and your right .. accidently took String/name out :(

So I need to test coercion to primitives to answer my initial confusion..

supertick, 2019-04-07:
primitives defined interface methods "work" but .. I'd say they are not optimial..
As to the rest ...
http://myrobotlab.org/content/servo-servocontrol-servocontroller-and-confusion

supertick, 2022-02-06:
worky ! - long ago fixed

closed 2022-02-06 by supertick