MyRobotLab/myrobotlab #492: LocalSpeech error

hairygael, 2019-04-13:
Testing with version 1.1.157
Becaus eNaturalReader is recently borked I am using LocalSpeech,
But it seems to add twice the same voice.
"2 was already added 2 female en_US" error
NaturalReader add a similar issue which I fixed in the java, but in the LocalSpeech jav file it's a bit more complex than just removing an extra line.

https://github.com/MyRobotLab/myrobotlab/blob/develop/src/main/java/org/myrobotlab/service/LocalSpeech.java#L137

supertick, 2019-04-13:
You want to post a no worky ? It might be related to how many voices are installed on your operating system.  For example, I only have 1.  Also - does this cause a problem with making or playing the speech ?
[image]

hairygael, 2019-04-13:
Hello Grog,
It doesn't cause a pb to play the speech.
It only throws that error which wasn't thrown under Manticore.
I have 6 voices in my PC.
I use the Mark voice, but there is a confusion between the voices numbers
of the pc and the voices name of localSpeech.
They are mixed up.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2019-04-13:
can you go to the {mrl dir}/tts/  - there should be a tts.exe on windows
do 
tts.exe -V  and copy paste what comes out
mine is the following 
[image]

hairygael, 2019-04-15:
Hello Grog,
Here is what comes out:
C:\MRL\myrobotlab.1.1.157\tts> .\tts.exe -V
0 Microsoft David Desktop - English (United States)
1 Microsoft Zira Desktop - English (United States)
2 Microsoft Hortense Desktop - French

Although it's missing some voices because I actually have this:
[image: Narrator.PNG]
Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2019-04-16:
Fix should be in build 158 - 
Since your testing I merged it to "develop" branch - you should be able to pick it up as the "latest"
This is the PR it came from - https://github.com/MyRobotLab/myrobotlab/pull/494
Usually kwatters and I review each other's pull requests - but since your going to test and "worky" is king and it was relatively small changes I merge it myself.

hairygael, 2019-04-16:
Hello,
Thanks Grog
Tested version 1.1.158
It has evolved but it is not solved from my side of view.
I have sent a no worky.
MRL still throws an error regarding the choice of voice no matter what I select in the InMoov configs:
Microsoft Hortense Desktop - French
 was already added  Microsoft Hortense Desktop - French
 null null
Plus no matter what I select as voice, I get always the same one which seems to be David.
So in the configs I tried:
VoiceName=David
VoiceName=0
VoiceName=Mark
VoiceName=1
VoiceName=2
VoiceName=Hortense

supertick, 2019-04-16:
I can't (although I've tried) to load more than one voice on my windows 7 system.  It's nuts - their are so many instructions, none of them work.
But I "might" have fixed the problem for you ...
VoiceName=David  would be the way to set a voice ...

moz4r, 2019-04-16:
In some cases we cannot use a standard pastern :

C:\mrl\tts>tts -V
0 Microsoft Hortense Desktop - French
1 Microsoft Paul Mobile - French (France)
**2 ScanSoft Katerina_Full_22kHz**
3 Microsoft Zira Desktop - English (United States)
4 Microsoft Julie Mobile - French (France)

In this case **2 ScanSoft Katerina_Full_22kHz** is not listed inside mrl.

We need to use the number as the voice name I think.
Or maybe the full string for voice name with default parameters for language, if the voice name don't follow the pattern. 
Or the full string in every cases but it can be ugly, but worky for "exotics" voice names.

supertick, 2019-04-18:
Good point Moz4r.
I changed AbstractSpeechSynthesis and AbstractSpeechSynthesis to also support voice keys which are all the parts of a voice. Now it doesn't matter if we parse it into pieces or not - what you see in the drop down can be used to setVoice("1 Microsoft Paul Mobile - French (France)").
If someone bothers to figure out the microsoft chaos - then the different parts are supported too, at the moment I just pulled it out (but left the French matching ;)

supertick, 2019-04-18:
Please test - verify this and we can close it... I tested with LocalSpeech (I can only get 1 voice on my machine & Polly)

moz4r, 2019-04-18:
I think the ability to chose between the key or label is really great . This would also solve the problems of backward compatibility because until today we used the key ( setVoice (X) ) for localSpeech

setVoice(Integer)
+
setVoice(String)

?

About the sample

1 Microsoft Paul Mobile - French (France)

I think the key is 1 and the label **Microsoft Paul Mobile - French (France)** but I can't use setVoice(1) anymore

supertick, 2019-04-19:
heh .. the key would be the entire "what you see" in the jcombo ..
but the fact there was a mis-understanding probably warrents having another "key" of integer too ;)

I'll add that as well

supertick, 2019-04-19:
All of these should work now

speech.setVoice("0 Microsoft Anna - English (United States)")  # full what you see in combo drop down
speech.setVoice("0") # string representation of voice provider
speech.setVoice(0)  # index of voice
speech.setLanguage("en")  # language of voice for at least French & English

supertick, 2019-04-19:
can this be closed ?

moz4r, 2019-04-19:
Thanks, we are stable we are WORKY !

closed 2019-04-19 by supertick

supertick, 2019-07-02:
found your update moz4r on localspeech branch ..
merged it into mine - tested - worky for me .. Thanks ! 
will flow into develop next pr i submit :)