MyRobotLab/myrobotlab #493: Minimal Non-Service attachable class that registers itself in the Registry

supertick, 2019-04-13:
With the new Attachable interface, there is the possibility of light weight classes which communicate and can accept or send messages to other full "services".

One thing that is currently lacking is the ability to "register" one of these light-weight classes.
With an unique entry in the registry, a encoder class could relay msgs or encoding data from a controller to a listener.

Runtime is keeping the service lifecycle structured, but for a class which does not have any of its own threads, is it necessary to start and stop it ?

kwatters, 2019-04-14:
I'm a little skeptical of this change.  What initial services/classes do you think will use this?  And why would you choose one of these light weight service wrappers instead of the normal service parent class?

supertick, 2019-04-15:
We chose not to make a Pin a service, yet an Encoder is very close to that level.  So, this is partially about finding where the boundary is.  Best practice to make a service vs supporting a few interfaces.

I'm also interested in what minimally is needed to provide communication between some class and the rest of the framework.  At the moment it means extending from Service.  The ServiceInterface is pretty large, and a significant amount of implementation is in Service.  What would be the "minimal" amount to support working correctly ?

I'm interested in the minimal definition, because if we were to make polyglot versions of mrl in python or javascript - it would be more clear on what an effort that would be.

Not to worry though, I don't intend changing any existing services.  I do intend to make some encoders possibly work without being services.  They will support the attach & encoder interfaces ... but finishing the refactor or Servos & InMoov is priority - I just put this ticket in to not forget.

supertick, 2019-04-19:
Ok .. did a little exploration work and ... ya .. TimerEncoder should be a service :)
Its still a good exercise to determine what minimally it takes to make a successful service, and what this means.  
It would also be good in the future to figure out how polyglot services can be easily and effectively 'attached' to Java-land ones.

For the TimeEncoder I found services needing attachment would refresh controllers (and potentially encodercontrol) and the refreshing of course comes from the Runtime registry ..  if its not part of the registry - lots of things become complicated.

So the design I'm now working with for TimeEncoder is similar to Pid except you don't need to "name" the pid instances as you do with Pid.  Instead, simply the TimeEncoder will associate the attaching service and the name'd instance of specific config for that service - will be created and associated in the attach(encoder) function..

closed 2019-04-19 by supertick