MyRobotLab/myrobotlab #495: Chatbot doesn't start new session

hairygael, 2019-04-16:
Version 1.1.158, InMoov-develop
Hello,
This issue was solved a few versions ago.
Chatbot doesn't start new session correctly when a new username is given. (John)
It doesn't create the predicates according to the name although you can see it executed in the prompt command.
<mrl><service>python</service><method>exec</method><param>i01.chatBot.startSession(u"John")</param></mrl> 
I didn't try to see if Face Recognizer is also not doing the job of .startSession(u"John") but I suspect that it is also out of work.

hairygael, 2019-04-17:
Hello
This is fixed with version 1.1.159. The session is now correctly started and the predicates are correctly setup.
Using inmoov-develop:
But now the whole process of initialisation of InMoov via START_INMOOV.bat when the robot would ask his name and your name to set a first session is busted.

supertick, 2023-04-02:
make a bdd ticket for the startup

closed 2023-04-02 by supertick