MyRobotLab/myrobotlab #513: Improve ACK support for MrlComm

kwatters, 2019-06-29:
mrlcomm has support to send an acknolwedgement for every message it receives.  

This ticket is to test the effects of the ACK enabled or disabled to see if it's still required.  If it is still required, we need to address the blocking wait for ack so that it's less intrusive and allow for a fast "re-sync" operation.

supertick, 2019-06-30:
I'm currently disabling it.  I have done testing with both acking and non-acking.
I believe the implementation minimally needs considerable revising.
One of my tasks was to get a new feature implemented I'll call re-synching.
This covers the scenario when mrl is running, but the arduino board browns out or resets for some other reason.   The feature "synchs" the state of mrl with the board, after communication has re-established.

It works surprisingly well.  I can play with servos, hit the reset button and immediately after the reset the servos are ready, additionally an error message is sent regarding the reset.  So the user is notified of the possible brownout problem, but not affected by it.

Acks are supposed to provide message quality, but their current implementation attempts to do this at the cost of freezing any communication.

This is not desirable.  A similar analogy would be your watching a movie, and it stops ..  the person running the projector says - sorry we had to stop the movie because a pixel was missing, we'll start it again in a hour ..   I'd prefer to be notified after the show a pixel was missing, and to watch uninterrupted.  ;)

kwatters, 2019-06-30:
So long as we don't overrun the serial port uart buffer on the arduino mega/uno we can probably get along without the acks.  but if we do some testing and periodically messages are being garbled, we will need to implement some sort of flow control when writing data to the arduino over a serial link.

kwatters, 2020-05-23:
I think this was all addressed with the better_acks2 branch/move.  I'm closing this ticket.  I'm relatively happy that acks are working and resync is working after the merge of that branch.

closed 2020-05-23 by kwatters