MyRobotLab/myrobotlab #514: Reattach all Arduino devices

supertick, 2019-06-29:
Recently the Arduino service has been enhanced with the capability to auto-magically re-attach to any Servo or HobbyServo service..  This is great, but all services that know how to attach to an Arduino should have this same functionality.

This will take some effort, as it means usually the detached devices need to re-initalize .. it might be quick or some parts may not be idempotent.  It will require further testing too.

kwatters, 2019-06-30:
Perhaps a default implementation would be to record the original attach msg and save that in the device summary.  so , when a re-sync occurs, it just plays back the original message that was used to attach.

supertick, 2019-12-14:
pins and servos correctly re-attach
moving this to ogre - and it will need to list other devices