MyRobotLab/myrobotlab #530: remove "bots" from ProgramAb path

supertick, 2019-07-13:
currently it looks like "bots" is always concatenated on the end of the root path in ProgramAb.
This prevents users from setting the root where they want .. because it always ends with "bots".
Contractually, the user should be able to set the root wherever they want and /{robotname} should come after
e.g.
if I set the path to "/home/gperry/someFolder"  the expectation is all bot folder be underneath
/home/gperry/someFolder/{botname}

kwatters, 2019-07-26:
This will require changes to ProgramAB.  Once those are ready, we'll update our dependency and get rid of the "bots" sub-directory.

kwatters, 2019-12-08:
I feel like this should be in nixie...  going to start on it shortly.  this does mean that the chat bot artifacts will need to likely change which directory they are in...  This will basically be to remove the "bots" sub directory for MRL purposes.

supertick, 2021-06-12:
I believe this is done (a long time ago)

closed 2021-06-12 by supertick