MyRobotLab/myrobotlab #573: Servo serialization issue

moz4r, 2019-11-17:
When servo.loadAndStart() is called, used to get things like last servo position, some things are broken. 

ERROR class org.myrobotlab.service.Arduino - getDeviceId could not find device 

I will set controller String to trancient inside servoAbstract, seem worky, to talk about.

supertick, 2019-11-21:
Hi Moz4r ..
Can you give me a sample script and/or instruction to replicate ?
Making the controller transient will take it out of serialization, and that in turn will break the webgui ServoGui and of course prevent transport from process to process unless get controller name methods are created.

moz4r, 2019-12-02:
Hi Captain ! sorry for the delay, this will break spacetime ( the second time of used, so if servo01.json exist ) 

```
virtualArduino = Runtime.start("virtualArduino", "VirtualArduino")
virtualArduino.connect("COM42")
arduino = Runtime.start("arduino","Arduino")
arduino.connect("COM42")
servo01 = Runtime.loadAndStart("servo01","Servo")
print servo01.getPos()
servo01.moveTo(42)
```

supertick, 2019-12-09:
I think its time we move past loading a big blob of json.  Getting back to a good state is usually more complex than just setting the values of a bunch of variables.

@kwatters - initially made some "calibration" code which dumped out a Python script - the script was minimal, because it started the system up normally, and didn't slam all the variable values in at once.

I continued with this idea, and now have org.myrobotlab.lang.LangUtils  ...   which will allow generation of code in order to get back to a "good" state.

I'm hopeful once this is demonstrated by configuration from the webgui - the user will have a nice clean, clear python script they can use for their inmoov

supertick, 2019-12-31:
current position is saved in /data/Servo/positions.json
Additionally, when we get to the export import python scripts relating to configuration - these values will be exported and 'set' on import

[image]

closed 2019-12-31 by supertick