MyRobotLab/myrobotlab #585: Strange behavior with text overlaps in webgui

hairygael, 2019-12-21:
**Describe the bug**
Strange behavior with text overlaps

**To Reproduce**
Just run the latest version 1.1.318 with a arduino and a servo, open the wegui

**Expected behavior**
Expecting no text overlap

View the effect:
[http://inmoov.fr/wp-content/uploads/2019/12/mrl.1.1.318_1.png](url)

**Desktop (please complete the following information):**
 - OS: windows 10 64
 - Browser firefox and chrome
 - Version firefox 71.0

hairygael, 2019-12-21:
Some other behaviors that seem odd for a newbie like me.
It seems the roll down menus are not showing.
I can't select the arduino or the pin when Chrome window is not in full size mode
http://inmoov.fr/wp-content/uploads/2019/12/mrl.1.1.318_2.png

Maybe some preset speed in roll down menufor newbie to know what is the minimum and maximum?
[image]

Maybe the "auto-enable on" and "auto-enable off" with a green and red color to show it is active or not?
Currently you don't know if "auto-enable on" means it is active or if you need to click to turn it on.

supertick, 2019-12-31:
Fixed on servo_pub_sub branch
https://www.youtube.com/watch?time_continue=3&v=WKdT6OXfyf4

closed 2019-12-31 by supertick