MyRobotLab/myrobotlab #587: Detach does not seem to work for Neopixel.

srayner, 2020-02-04:
When in the Neopixel service, clicking the 'Attach' button works ok. The text on the button changes to 'Detach'. However clicking the 'Detach' button does not seem to do anything.

I notice there is a TODO: in the neopixel code.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Neopixel'
2. Select Controller, pin, and no of pixels.
3. Click 'Attach'
4. Click 'Detach'

**Expected behavior**
Controller to detach, and button text to change back to 'Attach'

hairygael, 2020-02-07:
Hello,
Yes, this is a known issue on the Manticore version. (1.1.2693)
But the developers are not fixing Manticore anymore. I only fix the InMoov
errors which are located within the InMoov directory for the Manticore.
Hopefully Nixie version will have this fixed.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2022-02-06:
fixed in Nixie

closed 2022-02-06 by supertick