MyRobotLab/myrobotlab #588: Manticore: MrlComm will not compile

kg5kzh, 2020-02-08:
It used to compile not anymore have fresh copy of Manticore  tried raspi, win10 and win7 same error on all. see below.

Arduino: 1.8.11 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"
 
In file included from sketch\LinkedList.h:72:0,
 
                 from sketch\MrlComm.cpp:10:
 
sketch\LinkedList.cpp: In instantiation of 'LinkedList<T>::~LinkedList() [with T = Device*]':
 
sketch\MrlComm.cpp:29:18:   required from here
 
sketch\LinkedList.cpp:21:13: warning: ISO C++ forbids comparison between pointer and integer [-fpermissive]
 
   while(root!=false) {
 
         ~~~~^~~~~~~
 
sketch\LinkedList.cpp: In instantiation of 'LinkedList<T>::~LinkedList() [with T = Pin*]':
 
sketch\MrlComm.cpp:29:18:   required from here
 
sketch\LinkedList.cpp:21:13: warning: ISO C++ forbids comparison between pointer and integer [-fpermissive]
 
sketch\LinkedList.cpp: In instantiation of 'ListNode<T>* LinkedList<T>::getNode(int) [with T = Device*]':
 
sketch\LinkedList.cpp:182:29:   required from 'T LinkedList<T>::get(int) [with T = Device*]'
 
sketch\MrlComm.cpp:228:38:   required from here
 
LinkedList.cpp:53:10: error: cannot convert 'bool' to 'ListNode<Device*>*' in return
 
   return false;
 
          ^~~~~
 
sketch\LinkedList.cpp: In instantiation of 'ListNode<T>* LinkedList<T>::getNode(int) [with T = Pin*]':
 
sketch\LinkedList.cpp:170:29:   required from 'T LinkedList<T>::remove(int) [with T = Pin*]'
 
sketch\MrlComm.cpp:297:24:   required from here
 
LinkedList.cpp:53:10: error: cannot convert 'bool' to 'ListNode<Pin*>*' in return
 
exit status 1
cannot convert 'bool' to 'ListNode<Device*>*' in return
 
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
 
 
Thanks I know you have a lot going

ApiBiwan-InMoov, 2020-12-30:
seems to me that the latest version of the code (https://github.com/ivanseidel/LinkedList/)  is returning NULL instead of false

kwatters, 2020-12-31:
This issue was fixed on the "develop" branch.  If you use an older version of the Arduino IDE  ( 1.6.8 ), the IDE does not complain about this.  However, newer versions of the Arduino IDE are stricter when it comes to certain items that were 

Here's another thread about the same issue.  Line 53 needs to be updated/changed.

http://myrobotlab.org/content/myrobotlab-not-wurky-me

For reference:

https://github.com/MyRobotLab/myrobotlab/blob/develop/src/main/resources/resource/Arduino/MrlComm/LinkedList.cpp#L53

supertick, 2022-02-06:
nixie mrlcomm compiles

closed 2022-02-06 by supertick