MyRobotLab/myrobotlab #849: Service python Python can not be created

Gleis1234, 2021-06-01:
**Describe the bug**
Trying to install myrobotlab ends with an error:
15:32:10.346 [main] ERROR c.myrobotlab.framework.Service - runtime error could not create service python Python
15:32:10.347 [main] INFO  class org.myrobotlab.service.Runtime - attempting to invoke : python.execFile(./InMoov/InMoov.py )
The install script start-inmoov.sh ends up in a "runtime: /#" shell.
When going on and starting myrobotlab.jar there is again the python missing service error and no python script can be started.

**To Reproduce**
1. Unpack myrobotlab.1.0.2693.16.zip
2. moving out all files from direktory <home>/mrl/myrobotlab.1.0.2693.16 up one level to <home>/mrl .
3. starting start-inmoov.sh .
4. Error occurs.

**Expected behavior**
Installation finishes without error.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Kubuntu 20.04 LTS
 - Browser Firefox
 - Version 88.0.1

**Smartphone (please complete the following information):**
 no

**Additional context**

Gleis1234, 2021-06-02:
Adding the myrobotlab.log.
[myrobotlab.log](https://github.com/MyRobotLab/myrobotlab/files/6582445/myrobotlab.log)
Can anyone help me to understand the reason for the error?

supertick, 2021-06-02:
Hi @Gleis1234  - the problem is your running Java 12 with something which only supports up to Java 8
Short term solution - uninstall Java 12 and install Java 8.
Long term solution - we are working on Nixie version which is continually tested on Java 11 - http://myrobotlab.org/ , however, at this time is not ready for release.

closed 2021-06-02 by supertick