MyRobotLab/myrobotlab #868: Merge MimicSpeech into LocalSpeech

supertick, 2021-06-14:
#867 will give LocalSpeech the ability to use command templates to activate local TTS executable.

At the moment the PR is only for LocalSpeech Linux, and should be expanded in general for any of the operating systems.

"Perhaps" the mimic.exe from MimicSpeech should be bundled with LocalSpeech, and MimicSpeech removed. 
If mimic.exe stays, it should be bundled and deployed appropriately (not in the root mrl directory)

Cyber-One, 2021-06-14:
The service page can then hold a list of known worky templates for different TTS systems and the OS they are used on and even how to install them.

kwatters, 2021-06-14:
If we expect people to install it themselves, they will get it wrong and we will have to answer the question of why it doesn't work.

Cyber-One, 2021-06-14:
That's where documentation come in, if the instruction on how to do it are there and clear, then a lot will work it out.
If you limit it to just what is installed by MRL, then you limit what can be done with MRL and you need to have a solution for a number of different systems.
That seems like a lot more development work.

kwatters, 2021-06-14:
the template needs to be included in the cached wav file filename.  Otherwise, changing templates would be using the wrong cached file.

closed 2021-07-14 by supertick