MyRobotLab/myrobotlab #869: MRL Nixie download link doesn't work

adamzxtan, 2021-06-15:
Hi guys, I tried downloading MRL Nixie version from the github's page download section, but nothing happens. The only way I can get the MRL Nixie is at myrobotlab.org website, by clicking on the download link "latest 1.1.511" at the top side of shoutbox.
 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/MyRobotLab/myrobotlab
2. Scroll down to "Download the myrobotlab.jar"
3. Click on "latest Nixie 1.1.X"

**Expected behavior**
No download job will automatically start. 

**Desktop (please complete the following information):**
 - OS: Windows 10 (OS build:19042.1052)
 - Browser: Google Chrome (Ver 91.0.4472.106, 64bit)

**Additional context**
Add any other context about the problem here.

supertick, 2021-06-15:
Hello Adamzxtan,
We currently don't auto-publish after every build.  You are correct, it is available on the myrobotlab.org website ..
And that link in turn points to our Jenkins server - http://build.myrobotlab.org:8080/job/myrobotlab/job/develop/ 
Nixie technically, has not been released.

adamzxtan, 2021-06-16:
Hi Grog, thanks for the clarification. If that's the case, may I suggest to add a note at the download section of the github page? So that new beginners (like me) will know the correct place to get the release.

Also, I also found that the download link for the stable Manticore release doesn't have the "myrobotlab.jar" file. Is there something wrong with the link?
[image]

supertick, 2022-07-09:
Good point. Lots has changed - made this release for Nixie on Github
[image]

closed 2022-07-09 by supertick