MyRobotLab/myrobotlab #870: MRL Nixie can't be installed on Raspi 4B

adamzxtan, 2021-06-15:
**Describe the bug**
I downloaded the latest MRL Nixie (build 1.1.511), and tried installing on my Raspi 4B. However, it looks like the console is forever stuck at this line:

00:24:34.203 [main] INFO c.m.i.FileIO [FileIO.java:1034] looking for Resource framework/ivy.xml.template
00:24:34.208 [main] INFO c.m.f.r.Repo [Repo.java:222] writing file libraries/ivy.xml
00:24:34.209 [main] INFO c.m.f.r.Repo [IvyWrapper.java:428] cmd java -jar ..\..\ivy-2.4.0-4.jar -settings libraries/ivysettings.xml -ivy libraries/ivy.xml -retrieve libraries/jar/[originalname].[ext]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to myrobotlab.org, download the latest Nixie build (1.1.511) onto Raspi 4B
2. Create a new folder, I call mine "mrl"
3. Install Nixie in the folder by "java -jar myrobotlab.jar --install"
4. See error

**Expected behavior**
The console will be printing out a bunch of things at the beginning, and then it will stay stuck at this line:
`[main] INFO c.m.f.r.Repo [IvyWrapper.java:428] cmd java -jar ..\..\ivy-2.4.0-4.jar -settings libraries/ivysettings.xml -ivy libraries/ivy.xml -retrieve libraries/jar/[originalname].[ext]`

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Raspberry Pi OS (32bit) on Raspberry Pi 4B
 - Browser: Chromium

kwatters, 2021-06-15:
That's not an error.  You really just need to be patient on that line.  It's where all of the dependencies are resolved and MRL starts downloading them all.  you should see data being written into the ivy2 cache  (  /home/pi/.ivy2 ) so that it will be cached the next time you want to install.

supertick, 2021-06-15:
Also it might be worth noting ...  there's a possibility of it hanging forever if you have to go through a proxy.  I don't think the command honors system proxy settings ....

supertick, 2021-06-15:
I tried with debug level on and see a large amount of failures which should be coming from maven central ..
transitory maven central errors \o/ ?

	==== central: tried
	  https://repo.maven.apache.org/maven2/org/jxmpp/jxmpp-util-cache/0.4.2/jxmpp-util-cache-0.4.2.jar
		::::::::::::::::::::::::::::::::::::::::::::::
		::              FAILED DOWNLOADS            ::
		:: ^ see resolution messages for details  ^ ::
		::::::::::::::::::::::::::::::::::::::::::::::
		:: org.slf4j#slf4j-api;1.7.25!slf4j-api.jar(test-jar)
		:: org.slf4j#slf4j-api;1.7.25!slf4j-api.jar
		:: com.googlecode.json-simple#json-simple;1.1!json-simple.jar
		:: org.ejml#ejml-core;0.36!ejml-core.jar
		:: org.ejml#ejml-fdense;0.36!ejml-fdense.jar
		:: org.deepboof#learning;0.4!learning.jar
		:: org.codehaus.groovy#groovy-all;2.4.5!groovy-all.jar
		:: gov.nist.math#Jampack;1.0!Jampack.jar
		:: gov.nist.math#jama;1.0.3!jama.jar
		:: commons-collections#commons-collections;3.2.2!commons-collections.jar
		:: org.atmosphere#atmosphere-runtime;2.5.14!atmosphere-runtime.jar
		:: io.netty#netty-all;4.1.48.Final!netty-all.jar
		:: org.nd4j#jackson;1.0.0-M1!jackson.jar
		:: org.apache.commons#commons-compress;1.18!commons-compress.jar
		:: org.nd4j#nd4j-api;1.0.0-M1!nd4j-api.jar

		:: com.jakewharton.byteunits#byteunits;0.9.1!byteunits.jar

		:: org.nd4j#protobuf;1.0.0-M1!protobuf.jar

		:: com.github.oshi#oshi-core;3.4.2!oshi-core.jar

		:: net.java.dev.jna#jna-platform;4.3.0!jna-platform.jar

		:: org.threeten#threetenbp;1.3.3!threetenbp.jar

		:: it.unimi.dsi#fastutil;6.5.7!fastutil.jar

		:: com.twelvemonkeys.imageio#imageio-metadata;3.1.1!imageio-metadata.jar

		:: com.twelvemonkeys.imageio#imageio-tiff;3.1.1!imageio-tiff.jar

		:: com.twelvemonkeys.imageio#imageio-psd;3.1.1!imageio-psd.jar

		:: com.twelvemonkeys.imageio#imageio-bmp;3.1.1!imageio-bmp.jar

		:: org.bytedeco#opencv;4.5.1-1.5.5!opencv.jar

		:: org.bytedeco#opencv;4.5.1-1.5.5!opencv.jar

adamzxtan, 2021-06-16:
> That's not an error. You really just need to be patient on that line. It's where all of the dependencies are resolved and MRL starts downloading them all. you should see data being written into the ivy2 cache ( /home/pi/.ivy2 ) so that it will be cached the next time you want to install.

Yes, I thought so too. But I had been waiting for about 3 hours, and I know my internet speed isn't slow (3.1MB/s), so that seems to be very abnormal isn't it?

@supertick It looks like the only way to successfully install Nixie now is by clicking `install all services` inside the Webgui when Nixie is started. Here's what I did:

1. Go to myrobotlab.org, download the latest Nixie build (1.1.511) onto Raspi 4B
2. Create a new folder, I call mine "mrl"
3. Run Nixie in the folder by `java -jar myrobotlab.jar`
4. Nixie Webgui will start, navigate to "Runtime"
5. Click "install all services"

Now, the console in the `Runtime` should start printing a bunch of things. All services should start installing.

supertick, 2022-02-06:
verified on multiple raspis

closed 2022-02-06 by supertick