MyRobotLab/myrobotlab #885: Enhance ServoMixer

supertick, 2021-07-16:


supertick, 2021-07-16:
- Hide the Pose, Sequence and Sequence content with a hide/show button - so its out of the UI space
- 2 small arrows for 1 degree increments for each servo
- Tree of skeleton ? open up parts - Blender has a similar definition where a tree is searchable like a list, but still maintains its structure

supertick, 2021-07-16:
Current Servo Mixer - Servo with Find
[image]

Cyber-One, 2022-01-16:
Also the ability to control other inputs such as the X,Y and Z position of IK3D inputs would be very handy.
A number of poses could then be created to simulate walking or sitting where a defined sequence and directions of feet or arms needs to be planned.