MyRobotLab/myrobotlab #897: Random Page Loads

supertick, 2021-08-08:


hairygael, 2021-08-08:
This randomness happens everytime you simply start myrobotlab.jar.
I never get the randomness in any other case. Clicking on InMoov2 never got me a random UI as Kevin mentions in the shoutbox.
The default UI could be Runtime or Intro, Intro's the goal initially was to be the default starting UI.

hairygael, 2021-08-09:
Aaargh No!!! I just installed version 534 and now I get the randomness when clicking InMoov2 in the Intro page.

supertick, 2021-08-09:
Grrr ... fix it one way a problem fix it another way a problem ...
ok - I'll see if I can make both issues happy...

supertick, 2021-08-09:
Found the details of the problem - and I think in some places you've already adapted to what is a better UI experience.
What is happening:
    changeTab when told to go to a non existing tab would do nothing
    This behavior is not optimal - it adversely affects removing services or requests to go to tabs that do not exist
This was switched so that changeTab goes to runtime if the requested service does not exist.

In the InMoov2 UI - there are changeTab request right after request for creating new services.
e.g. startChatBot - sends the request to create the service, then immediately changeTab('chatBot')

This will not work - the creation of the service is asynchronous  - so going to the tab immediately after creating it (previously did nothing - because it didn't work) - now goes to runtime

The solution to this - is what you've done in chatbot panel - which is to create a button called "open" to allow the user to go to the appropriate tab, and additionally remove the changeTab('chatBot') code

I'll start updating the UI this way

supertick, 2021-08-09:
Was more simple than I thought - fixed in a way both problems are solved
committed directly to develop should be in build 1.1.535

closed 2021-08-11 by supertick