MyRobotLab/myrobotlab #931: servo slow to respond in webgui

mikehaskell, 2021-08-30:
**Describe the bug**
When using webgui servo service, the servo responds slowly after a couple movements

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'runtime'
2. Click on 'create arduino service' and configure for uno
3. Click on 'create servo service' and configure for servo
4. restart myrobotlab
5. goto servo menu
6. try moving servo several time.  it become slow to respond on my computer.
7. same test in swing gui results in full function
8. no errors

**Expected behavior**
servo should move without slow response.

**Screenshots**
i can create video if needed.

**Desktop (please complete the following information):**
 - OS: mac os x v10.11.6
 - Browser chrome 
 - Version 92.0.4515.159
2.4 GHz Intel Core 2 Duo
3 gig ram

**Additional context**
Add any other context about the problem here.
myrobotlab v 1.1.557
arduino uno r3
servo is ppwer hd Lw-20mg
webgui-client  goofy-marvin
Version 1.1.557
Branch develop
Platform x86.64.mac
Java Java HotSpot(TM) 64-Bit Server VM 11
Commit 3745840
Start Time 2021-08-30 14:23:48.184
Pid 615
Virtual false

kwatters, 2021-08-31:
Just to clarify a bit.. when you say  "restart myrobotlab" what do you mean?  are you just reloading your web browser?  Or are you actually stopping myrobotlab and starting it back up?

supertick, 2021-08-31:
I would tr y the latest.   I know of one thing which might have improved performance - the Time Encoder information is no longer being passed back to the webgui.  But this might be the root of the problem.

Please 
-  Download the latest  http://build.myrobotlab.org:8080/job/myrobotlab/job/develop/lastSuccessfulBuild/artifact/target/myrobotlab.jar
- in a new directory run  java -jar myrobotlab.jar --install
- Try the same activity as you did before and let us know the results

mikehaskell, 2021-08-31:
> Just to clarify a bit.. when you say "restart myrobotlab" what do you mean? are you just reloading your web browser? Or are you actually stopping myrobotlab and starting it back up?

Thanks for the question.

just the restart button in the right corner

mikehaskell, 2021-08-31:
Hi greg,
It got farther before it started to slow the response.  I will include the terminal info.  Sorry for the length.
there is probably more, but i thought that i might be enough

java:952] TIME-ENCODER SERVO_STOPPED - servo
17:33:42.576 [nioEventLoopGroup-3-2] ERROR c.m.a.Msg [Msg.java:2784] Ack not received, ack timeout!
17:33:42.576 [nioEventLoopGroup-3-2] WARN c.m.s.Arduino [Arduino.java:2197] Ack Timeout seen.  TODO: consider resetting the com port /dev/tty.usbmodem1D11, reconnecting and re syncing all devices.
17:33:44.395 [EventThread /dev/tty.usbmodem1D11] ERROR c.m.a.Msg [Msg.java:2501] Arduino->MRL error 97 rx sz errors
17:33:44.395 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 10 - 98 rx errors
17:33:44.395 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 3 - 99 rx errors
17:33:44.395 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 68 - 100 rx errors
17:33:44.395 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 2 - 101 rx errors
17:33:44.395 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 0 - 102 rx errors
17:33:44.395 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 18 - 103 rx errors
17:33:44.396 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 3 - 104 rx errors
17:33:44.396 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 16 - 105 rx errors
17:33:44.396 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 0 - 106 rx errors
17:33:44.399 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 1 - 107 rx errors
17:33:44.399 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 1 - 108 rx errors
17:33:45.578 [nioEventLoopGroup-3-2] ERROR c.m.a.Msg [Msg.java:2784] Ack not received, ack timeout!
17:33:45.578 [nioEventLoopGroup-3-2] WARN c.m.s.Arduino [Arduino.java:2197] Ack Timeout seen.  TODO: consider resetting the com port /dev/tty.usbmodem1D11, reconnecting and re syncing all devices.
17:33:45.579 [nioEventLoopGroup-3-2] INFO c.m.s.Servo [Servo.java:117] pos 134.0 output 134.0
17:33:45.580 [servo.servo.time-encoder] INFO c.m.s.a.AbstractServo [AbstractServo.java:941] TIME-ENCODER SERVO_STARTED - servo
17:33:45.982 [servo.servo.time-encoder] INFO c.m.s.a.AbstractServo [AbstractServo.java:952] TIME-ENCODER SERVO_STOPPED - servo
17:33:48.584 [nioEventLoopGroup-3-2] ERROR c.m.a.Msg [Msg.java:2784] Ack not received, ack timeout!
17:33:48.585 [nioEventLoopGroup-3-2] WARN c.m.s.Arduino [Arduino.java:2197] Ack Timeout seen.  TODO: consider resetting the com port /dev/tty.usbmodem1D11, reconnecting and re syncing all devices.
17:33:48.592 [EventThread /dev/tty.usbmodem1D11] INFO c.m.s.Arduino [Arduino.java:1753] publishServoEvent deviceId 1 event 0 currentPos 1854
17:33:48.594 [nioEventLoopGroup-3-2] INFO c.m.s.Servo [Servo.java:117] pos 136.0 output 136.0
17:33:48.594 [EventThread /dev/tty.usbmodem1D11] INFO c.m.s.Arduino [Arduino.java:1753] publishServoEvent deviceId 1 event 1 currentPos 1926
17:33:48.594 [servo.servo.time-encoder] INFO c.m.s.a.AbstractServo [AbstractServo.java:941] TIME-ENCODER SERVO_STARTED - servo
17:33:48.597 [EventThread /dev/tty.usbmodem1D11] INFO c.m.s.Arduino [Arduino.java:1753] publishServoEvent deviceId 1 event 0 currentPos 1926
17:33:48.598 [nioEventLoopGroup-3-2] INFO c.m.s.Servo [Servo.java:117] pos 138.0 output 138.0
17:33:48.601 [EventThread /dev/tty.usbmodem1D11] INFO c.m.s.Arduino [Arduino.java:1753] publishServoEvent deviceId 1 event 1 currentPos 1946
17:33:48.601 [EventThread /dev/tty.usbmodem1D11] INFO c.m.s.Arduino [Arduino.java:1753] publishServoEvent deviceId 1 event 0 currentPos 1946
17:33:48.602 [nioEventLoopGroup-3-2] INFO c.m.s.Servo [Servo.java:117] pos 140.0 output 140.0
17:33:48.605 [EventThread /dev/tty.usbmodem1D11] INFO c.m.s.Arduino [Arduino.java:1753] publishServoEvent deviceId 1 event 1 currentPos 1967
17:33:48.605 [EventThread /dev/tty.usbmodem1D11] INFO c.m.s.Arduino [Arduino.java:1753] publishServoEvent deviceId 1 event 0 currentPos 1967
17:33:48.606 [nioEventLoopGroup-3-2] INFO c.m.s.Servo [Servo.java:117] pos 141.0 output 141.0
17:33:48.610 [EventThread /dev/tty.usbmodem1D11] INFO c.m.s.Arduino [Arduino.java:1753] publishServoEvent deviceId 1 event 1 currentPos 1988
17:33:48.610 [EventThread /dev/tty.usbmodem1D11] INFO c.m.s.Arduino [Arduino.java:1753] publishServoEvent deviceId 1 event 0 currentPos 1988
17:33:48.613 [EventThread /dev/tty.usbmodem1D11] INFO c.m.s.Arduino [Arduino.java:1753] publishServoEvent deviceId 1 event 1 currentPos 1998
17:33:48.613 [nioEventLoopGroup-3-2] INFO c.m.s.Servo [Servo.java:117] pos 142.0 output 142.0
17:33:48.618 [EventThread /dev/tty.usbmodem1D11] INFO c.m.s.Arduino [Arduino.java:1753] publishServoEvent deviceId 1 event 0 currentPos 1998
17:33:48.619 [nioEventLoopGroup-3-2] INFO c.m.s.Servo [Servo.java:117] pos 143.0 output 143.0
17:33:48.625 [EventThread /dev/tty.usbmodem1D11] INFO c.m.s.Arduino [Arduino.java:1753] publishServoEvent deviceId 1 event 1 currentPos 2008
[... cut ...]
17:36:09.721 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 16 - 161 rx errors
17:36:09.722 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 0 - 162 rx errors
17:36:09.722 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 1 - 163 rx errors
17:36:09.722 [EventThread /dev/tty.usbmodem1D11] WARN c.m.a.Msg [Msg.java:2491] Arduino->MRL error - bad magic number 1 - 164 rx errors

mikehaskell, 2021-08-31:
could it be the mrlcomm  
how do i get the version number in the uno?

supertick, 2021-08-31:
I see at least 10 disconnect and reconnects...
Almost always this is a power issue.
Your Arduino is browning out (dying because not enough current)
Are you powering your servo from your Arduino ?  You should not be doing this if you are.
You need a separate power supply for the servos, and the Arduino and servo power supply need to be connected by ground.
The "timing out" in this case - is Arduino dies from not enough power, then reboots

mikehaskell, 2021-09-01:
the arduino is usb powered from imac.  the servo is powered from a 6v lantern battery. signal is pin 9 and ground for servo and arduino share ground from battery.  maybe imac usb is not correct voltage.  i will try loading myrobotlab on my laptop

kwatters, 2021-09-01:
It looks like you don't have the right version of MRLComm on your Arduino.  The latest version is 68.  It is not backwards compatible with Version 67.  You must upload the version of MRLComm.ino that comes with MRL in the resources/Arduino/MRLComm folder.

mikehaskell, 2021-09-01:
let me try that

mikehaskell, 2021-09-03:
Hi Greg and Kevin,
I connected a a 12v supply to the uno and nothing changed.  
I think this is the mrlcomm version that belongs with 1.1.563

version 68 load 18 sram 784 - 0 active pins 1 devices ts 1630632122885 heartbeat 1003
servo id:1
is that correct

supertick, 2021-09-03:
Right here
[image]

mikehaskell, 2021-09-05:
so i loaded the newest version 567 and correct mrlcomm for use on my windows 10 laptop.  all works well.  tried new version on imac with chrome or safari - results in same issue.  as previously stated i powered the uno separately than usb and also have a different ps for servo.  I believe the usb on the mac is the problem.  the commands from myrobotlab and the uno through usb are not working properly.  So I use a pc.  np

thanks for the help from the both of you.

supertick, 2022-02-06:
fixed - many optimizations for the ui at this point

closed 2022-02-06 by supertick