MyRobotLab/myrobotlab #938: Nixie LocalSpeech not working under MacOS

argoopen, 2021-09-29:
**Describe the bug**
Local speech is not working on MacOS using a recent release of Nixie (1.1.569).

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new LocalSpeech service
2. Navigate to the new LocalSpeech service and type in text for it to say
3. No audio is output, nor is any .aiff file created under audioFile/LocalSpeech

**Expected behavior**
The LocalSpeech service should output spoken TTS. On a Mac this is implemented via having the TTS binary /usr/bin/say output an AIFF file which is then played.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - I have tested this under various browsers and under two recent versions of MacOS (Catalina and Big Sur).

**Additional context**
The following log excerpt shows the problem:

`10:40:04.778 [nioEventLoopGroup-3-8] INFO c.m.s.a.AbstractSpeechSynthesis [AbstractSpeechSynthesis.java:568] Speech processing Test 1 2 3
10:40:04.784 [nioEventLoopGroup-3-8] INFO c.m.s.Runtime [Runtime.java:2330] execToString("say "test 1 2 3"-o /Users/pskipworth/dev/mrl/audioFile/LocalSpeech/fred/8183a5ce7c40b1247c8ddc622dc231a3.aiff null")
10:40:04.796 [nioEventLoopGroup-3-8] ERROR c.m.s.Runtime [Runtime.java:2394] execute threw
java.io.IOException: Cannot run program "say "test 1 2 3"-o /Users/pskipworth/dev/mrl/audioFile/LocalSpeech/fred/8183a5ce7c40b1247c8ddc622dc231a3.aiff": error=2, No such file or directory`

Whilst I am not a java dev, I am 95% sure there are a few issues here...

1. The MRL application is not correctly searching the user's PATH for the 'say' binary, which to my knowledge, has always lived under /usr/bin. Invoking it via the full path '/usr/bin/say', rather than just 'say', might be a possible solution here.
2. Even with 1. above being resolved, a space character needs to be added immediately before the "-o" command-line argument. 
3. I'm less certain of this, but I'm not sure the 'null' is required at the end of the command. In testing, adding the null actually results in the spoken word 'null' being added to the end of the output audio file.

In summary, here's an example of what is being invoked now:

`say "test 1 2 3"-o /Users/pskipworth/dev/mrl/audioFile/LocalSpeech/fred/8183a5ce7c40b1247c8ddc622dc231a3.aiff null`

...and here's what I suggest it should possibly  be changed to:

`/usr/bin/say "test 1 2 3" -o /Users/pskipworth/dev/mrl/audioFile/LocalSpeech`/fred/8183a5ce7c40b1247c8ddc622dc231a3.aiff null`

supertick, 2021-09-29:
Fixed in - pr #939

supertick, 2021-09-29:
Can you verify @argoopen  ?

argoopen, 2021-09-29:
Looks good @supertick . Thanks! Is there a way for me to test this now, or do I need to wait for it to be included in the next release?

Do I leave this bug open until I've been able to test it?

supertick, 2021-09-30:
yep - until its verified "worky"