MyRobotLab/myrobotlab #959: Bug in the operation of the PCF8574

Cyber-One, 2021-11-01:
I think there are some issue in the PCF8574 service that require the service to be simplified.

The way it's been created suggest the original authors thought it might work similar to the Arduino IO.
But it doesn't
The use of a direction register is not required and only adds to the confusion.
The method of setting a pin to an input is flawed and should be removed.
It uses one write register and one read register and the way it works, that all that is required.

The key to understanding the PCF8574 is in knowing how the IO pins work.
all 8 output pins are both an input and an output at the same time.
The trick is, the output are a very weak pull-up and a strong Pull-Down
That means, you can use it to drive the base of an LED, sinking current from and LED at up to 50mA where the LED is connected to a series resistor which is connected +5V
The week pull up can only source 400uA or 0.4 mA of current to ground.
In order to use the input to detect a push button or limit switch, Connect one end of the switch to the IO pin and the other to ground.
Write to the control register setting that pit high.
Read the input register, it will be high and until the button is pressed where it will then switch low.
Do not connect an input to the +5V rail via a button, as you can damage the PCA8574

In the service public void pinMode(int address, int mode) is not valid for this device
It tries to set a directionRegister which is also not valid for this device.
By the same logic enablePin and disablePin are also pointless.
public void write(String pin, int value) will work as will public int read(String pinName)
I wonder if that is also where the issues the HD44780 was having was starting from?
When I tested setting the pinMode to input, it was setting the output to low, this was never going to work on this device.

When you look at the JavaDocs, there are far more methods than are required to use the device.
All that is required is the ability to: 
Set and get the I2C Address
Set and get the I2C Bus
Set and get the I2C Controller
Attach and detach
Read Register
Write Register
Read pin
Write pin
Publish pin change.

To assist the user in using the device the docs will need to by updated to explain how the device should be used.
I will assist in this last point :-)
And in testing

Cyber-One, 2021-11-01:
An example of using the PCF8574 in MRL
https://github.com/Cyber-One/Fred_Inmoov/blob/main/Sample_KeyPad_PCF8574.py

Cyber-One, 2022-02-05:
In Version 1.1.721
There is provision for a scan rate of the IO device, but this does not appear to be working.
Nor is the current state updating in the WebGUI

supertick, 2022-11-13:
Related https://github.com/MyRobotLab/myrobotlab/issues/1177

supertick, 2022-11-13:
refer to : https://vimeo.com/770392660/924ef318b6 regarding problem

supertick, 2023-04-29:
fixed

closed 2023-04-29 by supertick