MyRobotLab/myrobotlab #985: Hd44780 - no worky !

supertick, 2021-12-12:
**Describe the bug**
Nothing shown on the LCD

**To Reproduce**
Steps to reproduce the behavior:
1. Start a raspi service
2. Start a Hd44780 service
3. attach them
4. nothing :(

**Expected behavior**
A worky screen

**Screenshots**

**Desktop (please complete the following information):**
 - raspi - raspian OS

**Additional context**

Ray Says :  https://discord.com/channels/887362610360643664/887362610360643667/919400215314853929

Cyber-One, 2021-12-17:
Process to start the HD44780 service on a Raspberry Pi is as follows.

Attach the HD44780 to the I2C pins on the Raspi
Start the RasPi service
Scan the I2C Bus to identify the correct address of the PCF8574
Start the PCF8574 service and attach it to the RasPi service with the address found from the I2C bus scan 
Start the hd44780 service and attach it to the PCF8574 service.
At least thats how it is supposed to work, but only random characters are displayed.

I suspect the issue is the HD44780 is not being initialised correctly.

How the HD44780 works

In this explanation we will be using the HD44780 in the 4 bit mode.
We will also be using the PCF8574 IO expander.

Just for your information, the HD44780 can operate in a 8 bit mode, 
but to save on hardware, and IO pins, a lot of systems choose to use the 4 bit mode.

There are 3 control line we need to use and we will use the 4 upper bidirectional data lines.
In most implementations of the PCF8574, the remaining IO line is used to turn the backlight on or off.

Lets first define the pins of the PCF8574 and how they are connected to the HD44780.
Pin		Description
0		Register Select, 0 = Instruction Register, 1 = Data Register.
1		Read/Write Register, 0 = Write, 1 = Read.
2		Enable Bit.
3		Backlight Bit.
4		Data Register Bit 4
5		Data Register Bit 5
6		Data Register Bit 6
7		Data Register Bit 7

Lets look at the HD44780.
The chip has two register that we can read from and write to:
0. The Instruction Register.
1. The Data Register.

Lets look at the instruction register first.

On powerup the display is in 8 bit mode.
We need it in 4 bit mode, data bits D0-D3 are not connected.
I have no idea what the lines 109 - 112 were trying to achieve.
We need to send direct to the PCF8574 register the following sequence
pcf8574.writeRigister(0b00100100 ) (Send the set register command for 4 bit)
pcf8574.writeRigister(0b00100000) ( Latch it in)
pcf8574.writeRigister(0b00100100)  (Send the set register command for 4 bit, need to do it twice to be sure 🙂  )
pcf8574.writeRigister(0b00100000) ( Latch it in)
loop
pcf8574.writeRigister(0b11110110) (Set it to read register)
HighNibble = pcf8574.readRegister()
pcf8574.writeRigister(0b11110010) (toggle the strobe)
pcf8574.writeRigister(0b11110110) (toggle the strobe)
LowNibble = pcf8574.readRegister()
repeat loop until bit 7 is low.
Now we are in 4 bit mode and the rest should work normally.
But we need to make sure when a command is sent to the Instruction register, that we check the status bit 7 as I did in the loop section

Hope that helps
Ray

Cyber-One, 2021-12-17:
A note on the PCF8574.
This I2C device is an IO expander, it has a single read register and a single write register.
Writing a 1 to a bit in the register will set the output to a logic 1
Writing a 0 to a bit in the register will set the output to a logic 0
The status register reports the current logic level of the IO pins.

This is where it can be confusing.
The PCF8574 has a very weak pull up when it its output is set to a logic 1 and can only source up to 400uA or 0.4mA.
And external source can pull this output back down to a logic 0, which the status register will return.
The PCF8574 has a strong pull down, it can sink around 40mA although it's recommended to stay below 25mA

In order to use the PCF8574 as an input device, you would write a logic 1 to the IO pin and allow the remote device to pull that pin down.

supertick, 2021-12-20:
Hd is fixed in https://github.com/MyRobotLab/myrobotlab/pull/1000

supertick, 2021-12-20:
running the service script makes it noWorky :disappointed:

supertick, 2021-12-20:
added synchronized to the main lcd writer and a 30ms delay which seems to improve the stability considerably

closed 2021-12-20 by supertick