MyRobotLab/myrobotlab #1013: WebGUI Config of the IK3D Service similar to other services.

Cyber-One, 2022-01-16:
**Is your feature request related to a problem? Please describe.**
The InverseKinematic3D service (IK3D Service) is a powerful and complex service with no User Interface in either SwingGUI or WebGUI.
What's more the IK3D Service is able to control multiple arms.

**Describe the solution you'd like**
A user interface that allows for a arm to be created or edited. (I'm assuming there will be a default arm in the defaultConfig)
The ability to add or remove links to the arm and edit the links as needed as the robot design is modified (make an arm longer or shorter or fine tune the parameters.)
The ability to attach the link to a servo/DIYservo, limits and current position can then be obtained from the servo service for the current link.
Also the ability to enable/disable the control of the servos on an arm. (This will allow other systems to control the arm.)

**Describe alternatives you've considered**
Coding in either Java or Python.
This method is not very NOOB user friendly.

**Additional context**
The ability to use a moveTo() function that selects the Arm, X, Y and Z axis target destination, number of steps, and time between steps would also be handy as part of the UI.
This service would be handy for each of the arms and the head of the Inmoov robot. as well as the SpotMicro Robot or any other robot that has a multi servo leg or Arm attached.

While it is possible to write your own forward and inverse kinematics for any arm up to 3  joint, beyond that does require the use of the IK3D service.  
The Inmoov Arm has 5 joints not including the fingers or the lower torso.
Inputs to the IK3D service should be accessible from the Servo Mixer or a service similar to the servoMixer created to support the WebGUI control of multiple arms.

The term "Arm" here is defined as a multi joint device that may have a hand, end-effector or foot attached to the end.

Ray Edgley

johnharland, 2022-01-17:
I would try and use this on my legs.
Currently: hips have pitch & yaw, knees have pitch, feet have pitch & roll,
so 5 degrees of freedom.
My last legs had 6, hips also had roll which I may add to my current legs.
All joints have angle sensors (AS5600) for feedback, and all motors are
controlled with RoboClaw bds.
The RoboClaw bds. use encoders for position and you can set current limits
on the motors.
So we need to support servos and DC motors.
Harland

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Cyber-One, 2022-01-17:
I'm surprised the Roboclaw service doesn't have support of moveTo() since it has an encoder input on it.
http://myrobotlab.org/service/RoboClaw

But yes once the moveTo() feature is added to that service, it could be included as well :-)
Ray

johnharland, 2022-01-17:
it does, both relative moves and absolute from 0
plus you can set speed, acceleration and deceleration on the move
  Drive M1 With Signed Speed And Acceleration
  Drive M1 with a signed speed and distance value
  Drive M1 With Signed Speed, Accel And Distance
  Drive M1 with signed Speed, Accel, Deccel and Position
there are a lot of commands
john