MyRobotLab/myrobotlab #1019: wrong assignment of neck servo in InMoov2

Gleis1234, 2022-01-25:
**Describe the bug**
When starting InMoov2_Full script the neck servo and the rollneck servo are both assigned to the same controller. In the config file(s) the neck servo is assigned to the right controller, the rollneck servo to the left controller

**To Reproduce**

1. Start InMoov2_Full config
2. try to address head.neck
3. try to address head.rollneck

**Expected behavior**
head.neck servo should be attached to i01.right controller.
head.rollneck should be attached to i01.left controller.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Kubuntu 20.4
 - Browser: Chrome
 - Version: 97.0.4692.99

**Smartphone (please complete the following information):**

**Additional context**
I can manually detach the neck servo and attach it to the right servo. Then the servo can be controlled by the slider.
[i01.head.neck.yml.txt](https://github.com/MyRobotLab/myrobotlab/files/7932070/i01.head.neck.yml.txt)
[i01.head.rollNeck.yml.txt](https://github.com/MyRobotLab/myrobotlab/files/7932071/i01.head.rollNeck.yml.txt)

hairygael, 2022-01-25:
Hello,
I just checked your issue, but they are properly addressed in the InMoov2_Full config.
Expected behavior:
head.neck servo is attached to i01.left controller.
[](https://github.com/MyRobotLab/InMoov2/blob/master/resource/InMoov2/config/InMoov2_Full/i01.head.neck.yml)
head.rollneck is attached to i01.right controller.
[](https://github.com/MyRobotLab/InMoov2/blob/master/resource/InMoov2/config/InMoov2_Full/i01.head.rollNeck.yml)

I also checked the InMoov2_Full script, and its also properly assigned.
[](https://github.com/MyRobotLab/InMoov2/blob/master/resource/InMoov2/system/startScripts/InMoov2FullStart.py#L388)
[](https://github.com/MyRobotLab/InMoov2/blob/master/resource/InMoov2/system/startScripts/InMoov2FullStart.py#L388)

Gleis1234, 2022-01-25:
Yes, you are right, concerning to the config files. But when I look to the real assignment in the running system, both servos are connected to to the left controller. That's what I am wondering about. To check I open the service i01.head.neck an i01.head.rollneck and look for the assigned controller.

Gleis1234, 2022-01-25:
Is your assignment rollneck to the right controller and neck to the left controller fix? Or can it be switched? I would prefer it the other way around.

Gleis1234, 2022-01-25:
I am testing with 1.1.701. My idea was, according to skeleton entry "RollNeckArduino", to define it as "left". This config seems to be correctly reflected in the config .yml files. There my intended controller is noted. But it seems that this does not be recognized in your start script. Is this script automatically generated? Then something in the generating process goes wrong.

Gleis1234, 2022-01-25:
If it is possible, can you explain how the handling of the config files works? There are the skeleton and service files in resource/InMoov2/config and the *.yml in the subdirs for each configuration. Then there are copies? in data/config .
In which should modifications be done, and what happens with them?

hairygael, 2022-01-25:
You should do your modifications in data/config.
Normally you shouldn't need to change things in resource/InMoov2/config
The copies are created into data when InMoov2 is launched for the first time.
Note, there is a script you can still adapt to your needs which is located in resource/InMoov2/system/startScripts/InMoov2FullStart.py . This script is launched when pressing the button "load Full Script".
However, this script is not meant to stay in the official release, but you can make a copy and launch it via the Python tab as well.
This script is only editable via a text editor and cannot be modified via UI saving buttons.

supertick, 2022-01-25:
Hello,
the resource/ directory is where services unzip the resources they came with - its not made to be modified by the user.
the data/ directory is owned by the user - and this is where configuration is loaded from

We have been developing the new configuration system for a while, but full configuration for InMoov has not been finalized.  It will need input from multiple people and some coordination.

But, you are correct in the goal being the user having the ability to update files in data/config/{config set} to their specific needs

supertick, 2022-01-25:
What is supposed to happen, when you get something working the way you want in the UI - you save it as a new configuration set 
[image]
But again, InMoov/2 has not been finalized. If your interested, you can follow more closely in our Discord channel - https://discord.gg/AfScp5x8r5

Gleis1234, 2022-01-25:
Thank you for your information. In the first step I will modify the FullStartSript until the full config is possible in the data/config files.
When I see this working I will close the issue.

hairygael, 2022-01-27:
Make sure you have set all virtual to false in the script:
Platform.setVirtual(False)
i01.setVirtual(False)
i01_right.setVirtual(False)
i01_left.setVirtual(False)
i01_controller3.setVirtual(False)
Make sure to comment:
#vi01_left_uart = Runtime.start('vi01.left.uart', 'Serial')
#vi01_right_uart = Runtime.start('vi01.right.uart', 'Serial')
#vi01_left = Runtime.start('vi01.left', 'VirtualArduino')
#vi01_right = Runtime.start('vi01.right', 'VirtualArduino')

Gleis1234, 2022-01-28:
Thank you for these hints. I did not see all of the necessary modifications.
But now I recognize errors with bad magic number in the communication to the arduinos. I tried with two versions of MrlComm, but both didn't work. One version will operate correctly with my manticore installation, the other one with the new nixie 1.1.701. Is there another version I have to use?

hairygael, 2022-01-28:
You will find the proper version of MRLComm.ino in
1.1.701/resource/Arduino/MRLComm/MRLComm.ino

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

Gleis1234, 2022-01-28:
That's exact i'm using when I work with 1.1.701.
Actually I have connected only two arduinos, one UNO for the right side, one MEGA for the left side. The controller3 i have commented out.
May it be a problem that I address the arduinos with Unix-notation /dev/serial/by-id/* . it's because the OS sometimes mixes up the assignment for the two Arduinos. This devs are  symbolic links to /dev/ttyACM* .
When I test the servos with the servo-function everything works as expected. Also when I call the InMoov2_Full config with "start+script".  Also with a modified config that I saved. In these cases only the assignment of the neck servos is not correct.
Error msgs from mrobotlab.log:
12:55:56.530 [i01.right] ERROR c.m.a.Msg [Msg.java:2790] Ack not received, ack timeout!
12:55:56.531 [i01.right] WARN c.m.s.Arduino [Arduino.java:2144] Ack Timeout seen.  TODO: consider resetting the com port /dev/serial/by-id/usb-Arduino__www.arduino.cc__0043_85036313230351618171-if00, reconnecting and re syncing all devices.
12:55:56.537 [i01.right] INFO c.m.s.Arduino [Arduino.java:467] ================ re-attaching i01.rightHand.ringFinger 2 5 ================
12:55:56.545 [i01.right] INFO c.m.s.Arduino [Arduino.java:467] ================ re-attaching i01.rightHand.wrist 3 7 ================
12:55:56.553 [i01.right] INFO c.m.s.Arduino [Arduino.java:467] ================ re-attaching i01.rightArm.bicep 4 8 ================
12:55:56.557 [i01.right] INFO c.m.s.Arduino [Arduino.java:467] ================ re-attaching i01.head.rollNeck 5 12 ================
12:55:56.562 [i01.right] INFO c.m.s.Arduino [Arduino.java:467] ================ re-attaching i01.rightArm.shoulder 8 10 ================
12:55:56.569 [i01.right] INFO c.m.s.Arduino [Arduino.java:467] ================ re-attaching i01.rightHand.index 9 3 ================
12:55:56.577 [i01.right] INFO c.m.s.Arduino [Arduino.java:467] ================ re-attaching i01.head.eyelidRight 7 24 ================
12:55:56.582 [i01.right] INFO c.m.s.Arduino [Arduino.java:467] ================ re-attaching i01.head.eyelidLeft 6 22 ================
12:55:56.590 [i01.right] INFO c.m.s.Arduino [Arduino.java:467] ================ re-attaching i01.rightHand.thumb 10 2 ================
12:55:56.594 [i01.right] INFO c.m.s.Arduino [Arduino.java:467] ================ re-attaching i01.rightArm.rotate 11 9 ================
12:55:56.602 [i01.right] INFO c.m.s.Arduino [Arduino.java:467] ================ re-attaching i01.rightArm.omoplate 12 11 ================
12:55:56.610 [i01.right] INFO c.m.s.Arduino [Arduino.java:467] ================ re-attaching i01.rightHand.pinky 13 6 ================
12:55:56.614 [i01.right] INFO c.m.s.Arduino [Arduino.java:601] Sync completed

These msgs are repeated over and over. May be it is a timing problem?

hairygael, 2022-01-28:
It seems your Arduino Uno is resseting.
Make sure you have:
i01_left.setBoardMega()
i01_right.setBoardUno()

Do you use the NervoBoard to power your servos? If not make sure you have a common ground for the Arduinos, servos and power supply.

Gleis1234, 2022-01-28:
The setBoard for the right Uno was not correct. I fixed it.
I power the servos over the nervo board with a separate accu. The arduinos are powered over the USB line from a USB hub with power supply. For the actual tests, to see the error msgs, I have not connected the power for the servos. 
Now I added the controller3 with a nano arduino and activated it in the script.
The error msgs comes as before. I add the actual log file.
I think the same effect happens, when I first started the InMoov config with only the start button. No errors appear, when I press the start+script button. What is the difference between the both ways?
[myrobotlab.log](https://github.com/MyRobotLab/myrobotlab/files/7959019/myrobotlab.log)

Gleis1234, 2022-01-28:
I Think I've got it to run. I disabled several components, which are not present (left arm, left shoulder, hands, etc.) Now it looks better. 
During the start the right shoulder made an extreme movement, which drives the omoplate piston out its holder. Is there a gesture at the start which could have caused that? Where can I look at gestures executed during start?

Thank you for the great help, and excuse me bothering you with my questions.

supertick, 2022-02-06:
Python should not be "necessary" for starting all of InMoov - it should be a combination of UI + configuration (in the form of yml files)
These details need to be worked out with the InMoov service - the python scripts functionality needs to be incorporated into the config, resolved with the ui, integrated into the service or removed

closed 2022-02-06 by supertick