MyRobotLab/myrobotlab #1035: Improvement to saving files in WebGUI

Cyber-One, 2022-02-04:
**Is your feature request related to a problem? Please describe.**
Back in the good old days of SwingGUI, you could open and save files with the python interface on the local machine, thats where MRL was running.
But under the WebGUI, we can be running the Web Client either on the computer running MRL or on another system entirely.
The WebGUI only allows you to open or save files to the machine where the Web Client is running.

**Describe the solution you'd like**
What we need is a way to save the files on the machine where MRL is running as well where the client is when not running on the localhost.

**Describe alternatives you've considered**
I have been using GitHub to transfer from the remote computer to the machine running MRL, it's cumbersome, but works.

**Additional context**
The ability to save both on the machine running MRL as well as the Web Clients localhost could be handy when debugging code on a test robot.

Ray

supertick, 2022-07-09:
This currently saves to machine running mrl.
[image]

closed 2022-07-09 by supertick

Cyber-One, 2022-07-09:
This only works when you are running the web client on the localhost.
It does not save it to the MRL machine if you are running the web client on a remote machine as I do 90% of the time.
I supress the starting of the web client on the localhost as the Raspi is not really powerful enough with chromium being such a resource hog.