MyRobotLab/myrobotlab #1038: Hd44780 Service

Cyber-One, 2022-02-05:
**Describe the bug**
After starting and attaching the PCF8574 service then starting the Hd44780 service, changing to the service page for the Hd44780, there are no devices listed that the service can be attached to.
There is also a runtime error happening when the service is being started.
The two bugs may be unrelated.

**To Reproduce**
Steps to reproduce the behaviour:
Using a Raspberry Pi 4, Buster OS, MRL Nixie 1.1.721
1. Using Runtime start the Raspberry Pi service and scan the I2C bus confirming the I2C device exists (Mine was on 0x27)
2. Using Runtime start the Pcf8574 service.
3. Change to the Pcf8574 service and attach it to the Raspi, bus 1 and selected I2C address of your LCD device 
4. Back to Runtime again, and start the Hd44780 Service. (Runtime error occurs here)
5. On the Hd44780 page attempt to select the Pcf8574 device you created previously.
6. You now have the error, there are no devices to choose from and you cant manually add one in.

it is possible to work around this using the Python command to manually attach the Hd4780 to the Pcf8574 service.
NOOBs would not know how to do this.

**Expected behavior**
The Hd44780 controller list should have any Pcf8574 service listed.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Raspbian
 - Browser Chromium
 - Version Buster

**Additional context**
This error also occurred.

```
12:41:26.068 [webgui-incoming-msg-queue] ERROR class org.myrobotlab.framework.Service - runtime error createAndStart(LCD, org.myrobotlab.service.Hd44780) java.lang.NullPointerException
12:41:26.128 [webgui-incoming-msg-queue] ERROR class org.myrobotlab.service.Runtime - createAndStart(LCD, org.myrobotlab.service.Hd44780) java.lang.NullPointerException
java.lang.NullPointerException: null
	at org.myrobotlab.service.Hd44780.writeRegister(Hd44780.java:282) ~[myrobotlab.jar:1.1.721]
	at org.myrobotlab.service.Hd44780.setBackLight(Hd44780.java:264) ~[myrobotlab.jar:1.1.721]
	at org.myrobotlab.service.Hd44780.load(Hd44780.java:346) ~[myrobotlab.jar:1.1.721]
	at org.myrobotlab.service.Runtime.start(Runtime.java:1886) [myrobotlab.jar:1.1.721]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.myrobotlab.service.WebGui$IncomingMsgQueue.run(WebGui.java:732) [myrobotlab.jar:1.1.721]
	at java.lang.Thread.run(Thread.java:829) [?:?]

```

Cyber-One, 2022-02-05:
The population of the attach list may be inhibited by the runtime error.
Clearing the runtime error in the WebGUI allows the list to be populated.

supertick, 2022-02-05:
npe was being thrown because - it loaded backlight config and tried to write to a pcf that wasn't there

closed 2022-02-05 by supertick