MyRobotLab/myrobotlab #1063: Latest Nixie Inmoov simulator?

AivisStud, 2022-03-04:
Hello,

 I am new to myrobotlab and I wanted to try out simulator (use it for some code development while Real Inmoov2 is under construction). I downloaded the latest Nixie version (from github link). and Tried to run simulator unfortunately I run in to issue that I do not know how to do it. I tried to follow tutorial (http://myrobotlab.org/content/nixie-inmoov2-sim-pdq). but in my case there are no buttons that let me start simulation.
[image]

What I am doing wrong in this case?
Was simulation startup sequcene changed since the release of mentioned tutorial?
should I download earlier version? (if yes then how should I delete the current version -just delete files from folder? something more)

Than you for your help in advance :)

hairygael, 2022-03-04:
Hello, 
Nixie is still under development some things are evolving. The button « start servos » is no longer of use. I suggest you launch the InMoov2_full.config, this will start the servo services and chatbot, this way you will have control over simulator.

AivisStud, 2022-03-04:
Hi Gael,

Thank you for information. I was able to run the simulator. thank you :)

other question is:
Can I use the simulator to check my own script how it would affect the robot. (I want to try and send some angles from python script to make robot move its arms).
is this possible and is there any information on how to get started?(could you point me to some documentation/tutorial where i could look in to this)

thank you for your response :)

hairygael, 2022-03-04:
Hello,
You can make your own python script and execute it via the python tab. 
Another solution is to set your script in the directory data/InMoov2/InMoovCustom.py
Every time you activate the button it will execute your script.
[image]
You can view and edit python gestures to understand how to make your own in resource/InMoov2/gestures

AivisStud, 2022-05-07:
Hi again,

Thank for the information. I played a bit with the simulation and now I would like to try my own code. but I have a question on how to send angles continuously? (like in this video of yours https://www.youtube.com/watch?v=q2zsUVFKZwY).
I have created a python code that calculates angles and I would like to see how simulator reacts. :) (before I try it on real thing).

could you provide some blueprint of code for that?

In my mind i would do something like this:

i01.startedGesture()
While loop
{
      Angles=getAngles(calculate angles)
      call (TestArms(angle1,angle2,angle3,angle4))
}
i01.finishedGesture()

would that work?

thank you in advance. :)

hairygael, 2022-05-11:
Hello,
You will need to use the IK3D service in Nixie.
Note that some things in Nixie can have bugs, so please report to
myrobotlab via github.
https://raw.githubusercontent.com/MyRobotLab/pyrobotlab/master/home/kwatters/DHParamsIK3D.py

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2022-07-08:
closing - simulator worky

closed 2022-07-08 by supertick