MyRobotLab/myrobotlab #1088: Multiple problems, ear in combination with chatbot and startup after shutdown

ggamehdtv, 2022-04-18:
Hi All,
Number 1: At first I tried setting up my controller and left arm, saved the config and shutdown mrl. After that I booted mrl loaded the config. And my controllers are shown as "off". No response or change when I click on "on". No error seen during this stage.

I expected to see everything on "on" after boot and loading the saved config via runtime. 

Number 2: When I setup chatbot I have sound and can talk to mrl using the chat function. Once I enable ear. It can hear me say things as I can see on the screen but it doesn't respond to that. Any things which should be changed to let it communicate with each other?

Number 3: After enabling torso, I got this error and mrl is completely stuck and requires a reboot to do anything. 
`01:40:32.678 [webgui-incoming-msg-queue] ERROR c.m.s.WebGui [WebGui.java:744] IncomingMessageQueue threw
java.lang.reflect.InvocationTargetException: null
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:566)
        at org.myrobotlab.service.WebGui$IncomingMsgQueue.run(WebGui.java:739)
        at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.lang.ClassCastException: class org.myrobotlab.service.Runtime cannot be cast to class org.myrobotlab.service.InMoov2Torso (org.myrobotlab.service.Runtime and org.myrobotlab.service.InMoov2Torso are in unnamed module of loader 'app')
        at org.myrobotlab.service.InMoov2.startTorso(InMoov2.java:1877)
        at org.myrobotlab.service.InMoov2.startTorso(InMoov2.java:1871)
        ... 6 common frames omitted`

OS:
 Windows 10, Minix pc on google chrome all 64bits. 
This is all with mrl 1.1.8.812.

Hopefully someone can help me with this one and solve the issues and get it up and running.

hairygael, 2022-04-18:
Hello,
1- this needs fixing, I have noticed this behavior as well, I believe it has to do with the recent modifications with peers.

2-try starting parts following this order and save you config: Mouth, ear, chatBot and save your config. Your config must have been saved in a wrong order, therefore things are not attached to each other properly.

3-probably again something related to peers modification. Hopefully a fix will be done soon.

From my iPhone, Gael

supertick, 2022-04-23:
1.  This is after pr 1093
https://youtu.be/X8PmeY1DLpM
Did a clean start with jar - started left & right - configured & connected - saved - restarted - found config and started ...
worky

supertick, 2022-07-08:
closing - fix was put in place - no activity or response since Apr 23

closed 2022-07-08 by supertick