MyRobotLab/myrobotlab #1095: No Connection Localhost (Nixie)

EGoll, 2022-05-03:
Hi everyone, 
I have a problem with Nixie (on Windows 11) and always get the message "Website is unreachable" after startup - localhost rejected the connection - ERR_Connection_Refused. After starting myrobotlab.bat the address is always called localhost:8888/#/tabs.

I have already uninstalled and reinstalled MRL, as well as Java (11.0.15) and Chrome. As a test, I also disabled the firewall, to no avail. I found the same problem in the forum (http://myrobotlab.org/content/localhost-refused-connect), but unfortunately there is no solution deposited.

Does anyone else have a tip for me on what I'm doing wrong? 

Thank you for your help.
Endric
[myrobotlab.log](https://github.com/MyRobotLab/myrobotlab/files/8610004/myrobotlab.log)

supertick, 2022-05-03:
Ah, I see the problem.
You may have installed Java 11.0.15 but your running Java 18
```

14:23:22.682 [main] INFO c.m.s.Runtime [Runtime.java:2243] java.specification.version [18]
14:23:22.682 [main] INFO c.m.s.Runtime [Runtime.java:2253] file.encoding [UTF-8]
14:23:22.683 [main] INFO c.m.s.Runtime [Runtime.java:2254] Charset.defaultCharset() [UTF-8]
14:23:22.684 [main] INFO c.m.s.Runtime [Runtime.java:2255] user.language [de]
14:23:22.684 [main] INFO c.m.s.Runtime [Runtime.java:2256] user.country [DE]
14:23:22.685 [main] INFO c.m.s.Runtime [Runtime.java:2257] user.variant []
14:23:22.685 [main] INFO c.m.s.Runtime [Runtime.java:2261] java.home [C:\Program Files\Java\jdk-18.0.1]
```
I'd recommend removing the Java 18 version.
If you want to run multiple versions of Java you could read through this - http://myrobotlab.org/content/how-run-java-11-and-java-8-same-machine

Cyber-One, 2022-05-03:
I was under the impression the MRL was targeted at Java 11 but was known to be working up to java 13 with things not working after that version.
I can only test up to Windows 10 with Java 12

supertick, 2022-07-08:
closed - incorrect Java version

closed 2022-07-08 by supertick