MyRobotLab/myrobotlab #1104: RasPi MPU6050 - Serialization Error

supertick, 2022-06-13:
Cyber found this 
https://discord.com/channels/887362610360643664/887362610360643667/985726201878446150

tCommitUserEmail":"kwatters@kmwllc.com","GitCommitUserName":"Kevin Watters","GitDirty":"false","GitRemoteOriginUrl":"https://github.com/MyRobotLab/myrobotlab.git","GitTags":"","Implementation-Version":"1.1.830","Main-Class":"org.myrobotlab.service.Runtime","Major-Version":"1.1.830","Manifest-Version":"1.0"},"shortCommit":"20fed2a"}}) - 1655076484473
09:28:04.560 [webgui-incoming-msg-queue] INFO c.m.f.Registration [Registration.java:47] creating registration for analog1@violet-rosie - org.myrobotlab.service.Ads1115
09:28:04.568 [webgui-incoming-msg-queue] INFO c.m.f.Registration [Registration.java:47] creating registration for intro@violet-rosie - org.myrobotlab.service.Intro
09:28:04.572 [webgui-incoming-msg-queue] INFO c.m.f.Registration [Registration.java:47] creating registration for mpu6050A@violet-rosie - org.myrobotlab.service.Mpu6050
09:28:04.596 [webgui-incoming-msg-queue] INFO c.m.f.Registration [Registration.java:47] creating registration for python@violet-rosie - org.myrobotlab.service.Python
09:28:04.600 [webgui-incoming-msg-queue] INFO c.m.f.Registration [Registration.java:47] creating registration for raspi@violet-rosie - org.myrobotlab.service.RasPi
09:28:04.612 [webgui-incoming-msg-queue] INFO c.m.f.Registration [Registration.java:47] creating registration for runtime@violet-rosie - org.myrobotlab.service.Runtime
09:28:04.616 [webgui-incoming-msg-queue] INFO c.m.f.Registration [Registration.java:47] creating registration for security@violet-rosie - org.myrobotlab.service.Security
09:28:04.623 [webgui-incoming-msg-queue] INFO c.m.f.Registration [Registration.java:47] creating registration for webgui@violet-rosie - org.myrobotlab.service.WebGui
11:35:40.811 [mpu6050A-publisher] INFO c.m.s.Mpu6050 [Mpu6050.java:326] Resetting MPU6050...
11:35:40.821 [mpu6050A-publisher] ERROR c.m.l.Logging [Logging.java:19] java.io.IOException: Remote I/O error
	at java.base/java.io.RandomAccessFile.writeBytes(Native Method)
	at java.base/java.io.RandomAccessFile.write(RandomAccessFile.java:559)
	at com.pi4j.io.i2c.impl.I2CBusImpl.lambda$writeBytesDirect$5(I2CBusImpl.java:179)
	at com.pi4j.io.i2c.impl.I2CBusImpl.runBusLockedDeviceAction(I2CBusImpl.java:261)
	at com.pi4j.io.i2c.impl.I2CBusImpl.writeBytesDirect(I2CBusImpl.java:178)
	at com.pi4j.io.i2c.impl.I2CDeviceImpl.write(I2CDeviceImpl.java:104)
	at org.myrobotlab.service.RasPi.i2cWrite(RasPi.java:347)
	at org.myrobotlab.service.Mpu6050.I2CdevReadBytes(Mpu6050.java:4287)
	at org.myrobotlab.service.Mpu6050.I2CdevReadByte(Mpu6050.java:4226)
	at org.myrobotlab.service.Mpu6050.I2CdevWriteBit(Mpu6050.java:4386)
	at org.myrobotlab.service.Mpu6050.reset(Mpu6050.java:3310)
	at org.myrobotlab.service.Mpu6050.dmpInitialize(Mpu6050.java:327)
	at org.myrobotlab.service.Mpu6050$OrientationPublisher.run(Mpu6050.java:82)
	at java.base/java.lang.Thread.run(Thread.java:829)

11:35:40.846 [mpu6050A-publisher] ERROR c.m.f.Service [Service.java:1905] raspi error i2c could not read
11:35:40.923 [mpu6050A-publisher] ERROR c.m.s.RasPi [RasPi.java:321] i2cRead threw
java.io.IOException: Remote I/O error
	at java.base/java.io.RandomAccessFile.readBytes(Native Method)
	at java.base/java.io.RandomAccessFile.read(RandomAccessFile.java:406)
	at com.pi4j.io.i2c.impl.I2CBusImpl.lambda$readBytesDirect$1(I2CBusImpl.java:150)
	at com.pi4j.io.i2c.impl.I2CBusImpl.runBusLockedDeviceAction(I2CBusImpl.java:261)
	at com.pi4j.io.i2c.impl.I2CBusImpl.readBytesDirect(I2CBusImpl.java:150)
	at com.pi4j.io.i2c.impl.I2CDeviceImpl.read(I2CDeviceImpl.java:190)
	at org.myrobotlab.service.RasPi.i2cRead(RasPi.java:318)
	at org.myrobotlab.service.Mpu6050.I2CdevReadBytes(Mpu6050.java:4288)
	at org.myrobotlab.service.Mpu6050.I2CdevReadByte(Mpu6050.java:4226)
	at org.myrobotlab.service.Mpu6050.I2CdevWriteBit(Mpu6050.java:4386)
	at org.myrobotlab.service.Mpu6050.reset(Mpu6050.java:3310)
	at org.myrobotlab.service.Mpu6050.dmpInitialize(Mpu6050.java:327)
	at org.myrobotlab.service.Mpu6050$OrientationPublisher.run(Mpu6050.java:82)
	at java.base/java.lang.Thread.run(Thread.java:829)

Cyber-One, 2022-06-13:
Something I discovered later, was the I2C bus had been disconnected from the MPU6050.
This could have caused an issue with communications from the computer to the MPU6050, but this should fail gracefully, not crash the whole system.

supertick, 2022-06-13:
I agree