MyRobotLab/myrobotlab #1106: PIR sensor no worky worky

jwkelly49, 2022-07-04:
x86.64.windows 1.1.831

All Servos work (right & left NervoBoards)
PIR on right NervoBoard

i01 -- goto sensors pir graphic ON/OFF button appears in the OFF position with no text.
activate pir -- button moves right turns blue settings button appears. Milliseconds later button returns left and grey with no text + settings button disappears.

Left column i01 pir appears (click on it)
Grey ball poll rate 1hz
set controller to i01.right pin=23
click attach -- and the enable button appears
click enable -- Grey ball + command prompt poll at 1 second interval with correct COMM port INFO c.m.s.Pir onPin pin=d23 value=0 (every 5-6 seconds value=1)
toggling enable/disable starts/stops polling
click detach -- (makes controller + pin assignment available)
Command prompt shows 
INFO c.m.s.Arduino [Arduino.java:670] i01.right detaching i01.pir
INFO c.m.s.Arduino [Arduino.java:684] detaching device i01.pir
ERROR c.m.s.Arduino [Arduino.java:1002] getDeviceId could not find device i01.pir

goto i01.right -- PIR is not listed with arm, stomach, hand, neck, eye

return to i01 sensor screen (pir shown in OFF position no text)
click button ON a multitude of times -- settings button appears/disappears as ON/OFF returns to OFF. No change in i01.pir object settings. No change in Command prompt polling. No change in i01.right controller settings.

i01.right O-scope has graph labeled MAX 1 D23 23 -- AVG 0 -- MIN 0  straight line (flat graph) at 0
i01.right pins D23 read + input

checked i01.left PIR not listed O-scope shows no chart for D23

Cyber-One, 2022-10-09:
Testing with version 1.1.974.
The green/grey indicator follows the state on the physical input on the Arduino.
Tested with Arduino Nano running MRL from Windows 10.
Not running the Inmoov service, just the Arduino Service and the PIR service with the PIR connected to D7.

hairygael, 2022-10-09:
Testing in version 1.1.974
In InMoov2 service starting the i01.right arduino and then the i01.pir. All worky on Windows 10

closed 2022-10-11 by supertick