MyRobotLab/myrobotlab #1126: Line endings should be consistent

holgerfriedrich, 2022-07-25:
Currently, text files (java, etc) use a different line endings (CR/CRLF), and some files contain a mix of both.
Working on #1121, I have trouble setting up spotless. The tool always tries to reformat nearly half of the codebase :-)

For java files, there are ~800 out of 1300 files using CRLF style.
Typically the convention is to use LF in the git repo and let git automatically decide on checkout depending on the target platform.

My recommendation would be to use `text=auto` feature in `.gitattributes` file to enable automatic conversion. However, this does not help for the files in the repo using CRLF format. It would require **a huge commit** fixing all the files, and basically **rendering "git blame" useless**, as this would always point to this commit. I see not other option, as rewriting the commit history is not a good way for a public repo anyway.

Any ideas on this, @AutonomicPerfectionist ?

AutonomicPerfectionist, 2022-07-26:
@holgerfriedrich line endings and messing with git are not my forte, unfortunately. I suggest you join the discord (https://discord.gg/AfScp5x8r5) to talk with the more experienced developers or ping them here (supertick, kwatters, etc.)

supertick, 2024-06-15:
Haven't seen this until now.
It's relatively stable and to my knowledge there aren't any large development tasks that have been branched off currently.
So, now would be the time to do it.
We've had major commits like this in the past when we processed the whole project under eclipse.
It's preferable to put all files in one format, and better if we can keep it that way.  "git blame" will be come useful again at some point.

supertick, 2024-06-15:
https://github.com/MyRobotLab/myrobotlab/pull/1427
If the auto-elves pass the build it will get merged

supertick, 2024-06-15:
merged

closed 2024-06-15 by supertick