MyRobotLab/myrobotlab #1137: MavenWrapper does not select correct scope of dependency if it's found at a different scoep earlier

AutonomicPerfectionist, 2022-07-31:
**Describe the bug**
When MavenWrapper generates the pom.xml, it loops over all of the services to find their dependencies. It will then determine what their scope should be by looking at the `includeServiceInOneJar()` property. If it finds a dependency that was already written, it will skip the dependency. However, if the first one had a different scope, then it will not be updated to match the scope of the skipped one.

**To Reproduce**
Steps to reproduce the behavior:
1. Add Jackson dependencies to RuntimeMeta
2. Generate pom.xml with MavenWrapper
3. Observe the jackson dependencies still listed as `provided` due to MQTT being processed first

**Expected behavior**
If a dependency that was already written is found but with a higher scope, then the scope should be updated.

supertick, 2022-08-01:
I think I'm starting to understand.
its not a higher scope - you just mean specifically since MQTT's dependencies collided with Runtime, they were stolen away.

If your intent is to package the deps in a build with Runtime, they should simply be removed from declaring them elsewhere like MqttMeta etc..

AutonomicPerfectionist, 2022-08-01:
I suppose that's a simple fix, but then if we ever remove Jackson from the framework stuff it would break all the services. I found where the issue lies (MavenWrapper in createPOM()) and it should be pretty easy to fix, just need a simple preprocessing stage to loop over dependencies and remove duplicates, keeping the one with the highest priority scope (i.e. packaged-in is higher priority than provided), then the writing stage doesn't have to worry about duplicates

PR #1138 (2022-09-10): Make POM generation respect includeInOneJar when duplicate dependencies exist
Fixes #1137 by adding a preprocessing stage to scan for duplicate dependencies and to select the one that requires inclusion in the final jar as the one to put in the pom.xml file.
closed 2022-09-10 by supertick