MyRobotLab/myrobotlab #1233: jMonkeyEngine is not working on Mac

YaowenGuo, 2023-01-02:
**When use jMonkeyEngine simulator to run InMoov, It crashed.**

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'i01'  InMoov page.
2. Click on 'Virtualize services' + 'Custom Script' + 'simulator'
3. It crashed.

**Expected behavior**
Start jMonkeyEngine simulator to run InMoov simulation on mac.

**Screenshots**
[image]
[image]

**Desktop :**
 - OS: Mac OS 13.0.1
 - Browser: Chrome  Version 108.0.5359.124 (Official Build) (x86_64)

**Error log**
```
22:26:17.528 [webgui-incoming-msg-queue] INFO c.m.s.JMonkeyEngine [JMonkeyEngine.java:1517] i01.simulator.onRegistered(efficient-johnny5 i01.mouth.audioFile org.myrobotlab.service.AudioFile)
22:26:17.528 [webgui-incoming-msg-queue] INFO c.m.s.JMonkeyEngine [JMonkeyEngine.java:1517] i01.simulator.onRegistered(efficient-johnny5 i01.mouth org.myrobotlab.service.MarySpeech)
22:26:17.528 [webgui-incoming-msg-queue] INFO c.m.s.JMonkeyEngine [JMonkeyEngine.java:1517] i01.simulator.onRegistered(efficient-johnny5 i01.simulator org.myrobotlab.service.JMonkeyEngine)
22:26:17.528 [webgui-incoming-msg-queue] INFO c.m.s.JMonkeyEngine [JMonkeyEngine.java:1517] i01.simulator.onRegistered(efficient-johnny5 i01.torso.lowStom org.myrobotlab.service.Servo)
22:26:17.529 [webgui-incoming-msg-queue] INFO c.m.f.Service [Service.java:1885] subscribe [i01.torso.lowStom/publishEncoderData ---> i01.simulator/onEncoderData]
22:26:17.529 [webgui-incoming-msg-queue] INFO c.m.s.JMonkeyEngine [JMonkeyEngine.java:1517] i01.simulator.onRegistered(efficient-johnny5 i01.torso.midStom org.myrobotlab.service.Servo)
22:26:17.529 [webgui-incoming-msg-queue] INFO c.m.f.Service [Service.java:1885] subscribe [i01.torso.midStom/publishEncoderData ---> i01.simulator/onEncoderData]
22:26:17.529 [webgui-incoming-msg-queue] INFO c.m.s.JMonkeyEngine [JMonkeyEngine.java:1517] i01.simulator.onRegistered(efficient-johnny5 i01.torso.topStom org.myrobotlab.service.Servo)
22:26:17.530 [webgui-incoming-msg-queue] INFO c.m.f.Service [Service.java:1885] subscribe [i01.torso.topStom/publishEncoderData ---> i01.simulator/onEncoderData]
22:26:17.530 [webgui-incoming-msg-queue] INFO c.m.s.JMonkeyEngine [JMonkeyEngine.java:1517] i01.simulator.onRegistered(efficient-johnny5 i01.torso org.myrobotlab.service.InMoov2Torso)
22:26:17.531 [webgui-incoming-msg-queue] INFO c.m.s.JMonkeyEngine [JMonkeyEngine.java:1517] i01.simulator.onRegistered(efficient-johnny5 i01 org.myrobotlab.service.InMoov2)
22:26:17.531 [webgui-incoming-msg-queue] INFO c.m.s.JMonkeyEngine [JMonkeyEngine.java:1517] i01.simulator.onRegistered(efficient-johnny5 intro org.myrobotlab.service.Intro)
22:26:17.531 [webgui-incoming-msg-queue] INFO c.m.s.JMonkeyEngine [JMonkeyEngine.java:1517] i01.simulator.onRegistered(efficient-johnny5 python org.myrobotlab.service.Python)
22:26:17.531 [webgui-incoming-msg-queue] INFO c.m.s.JMonkeyEngine [JMonkeyEngine.java:1517] i01.simulator.onRegistered(efficient-johnny5 runtime org.myrobotlab.service.Runtime)
22:26:17.532 [webgui-incoming-msg-queue] INFO c.m.s.JMonkeyEngine [JMonkeyEngine.java:1517] i01.simulator.onRegistered(efficient-johnny5 security org.myrobotlab.service.Security)
22:26:17.532 [webgui-incoming-msg-queue] INFO c.m.s.JMonkeyEngine [JMonkeyEngine.java:1517] i01.simulator.onRegistered(efficient-johnny5 webgui org.myrobotlab.service.WebGui)
22:26:17.536 [i01] INFO c.m.s.InMoov2 [InMoov2.java:221] i01.simulator@efficient-johnny5 created
22:26:17.536 [i01] INFO c.m.s.InMoov2 [InMoov2.java:95] i01.simulator started
22:26:17.536 [python] INFO c.m.s.Python [Python.java:478] exec(String)
from org.myrobotlab.framework import Platform
from org.myrobotlab.service import Runtime
from org.myrobotlab.framework import Service
from org.myrobotlab.service import JMonkeyEngine
i01_simulator = Runtime.getService("i01.simulator")

22:26:17.541 [python.interpreter.42] INFO c.m.s.Python [Python.java:171] script completed
22:26:17.541 [python.interpreter.42] INFO c.m.s.Python [Python.java:596] finishedExecutingScript
Jan 02, 2023 10:26:17 PM com.jme3.system.JmeDesktopSystem initialize
INFO: Running on jMonkeyEngine 3.3.2-stable
 * Branch: HEAD
 * Git Hash: 1a05e3f
 * Build Date: 2020-04-27
2023-01-02 22:26:17.914 java[55416:2022194] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007ff81a53843b __exceptionPreprocess + 242
	1   libobjc.A.dylib                     0x00007ff81a087e25 objc_exception_throw + 48
	2   CoreFoundation                      0x00007ff81a5605d6 _CFBundleGetValueForInfoKey + 0
	3   AppKit                              0x00007ff81d607161 -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 307
	4   AppKit                              0x00007ff81d5f3f4f -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1247
	5   AppKit                              0x00007ff81d5f3a69 -[NSWindow initWithContentRect:styleMask:backing:defer:] + 42
	6   libglfw.dylib                       0x000000013983dbc9 libglfw.dylib + 68553
	7   libglfw.dylib                       0x0000000139837416 libglfw.dylib + 42006
	8   ???                                 0x00000001159a86b0 0x0 + 4657415856
	9   ???                                 0x00000001159a29d0 0x0 + 4657392080
)
libc++abi: terminating with uncaught exception of type NSException
./myrobotlab.sh: line 66: 55416 Abort trap: 6           "${JAVA}" ${JAVA_OPTIONS} -cp ${CLASSPATH} org.myrobotlab.service.Runtime --from-launcher --log-level info -s webgui WebGui intro Intro python Python
0
```

YaowenGuo, 2023-01-10:
A temporary solution：Start jMonkeyEngine at start. 

edite: src/main/java/org/myrobotlab/service/JMonkeyEngine.java
```
-      mainThread = new Thread() {
-        @Override
-        public void run() {
-          app.start();
-        }
-      };
-
-      mainThread.start();
+      app.start();
```
add `-XstartOnFirstThread` and  `i01.simulator JMonkeyEngine` to `myrobotlab.sh`

```
-JAVA_OPTIONS="-Djava.library.path=libraries/native -Djna.library.path=libraries/native -Dfile.encoding=UTF-8"
+JAVA_OPTIONS="-Djava.library.path=libraries/native -Djna.library.path=libraries/native -Dfile.encoding=UTF-8 -XstartOnFirstThread"
...

-"${JAVA}" ${JAVA_OPTIONS} -cp ${CLASSPATH} org.myrobotlab.service.Runtime --from-launcher --log-level info -s webgui WebGui intro Intro python Python
+"${JAVA}" ${JAVA_OPTIONS} -cp ${CLASSPATH} org.myrobotlab.service.Runtime --from-launcher --log-level info -s webgui WebGui intro Intro python Python i01.simulator JMonkeyEngine
```
Compile from source and run it.

supertick, 2023-01-10:
Hello YaowenGuo,
Ah, Mac must run it off the main thread running the jvm .. I see.
:thinking:  The  -XstartOnFirstThread is fine, but some users may not want to start the simulator immediately, or at all.
Can you think of any way to do it "without" starting JMonkeyEngine immediately ?
with only this ?
`-s webgui WebGui intro Intro python Python `

supertick, 2024-01-10:
no activity

closed 2024-01-10 by supertick