MyRobotLab/myrobotlab #1237: Web gui not working

snnusta, 2023-01-03:
10:20:44.145 [main] INFO c.m.f.Platform [Platform.java:270] git loading properties from jar /C:/MRL/myrobotlab-1.1.1050/myrobotlab.jar
10:20:44.267 [main] INFO c.m.s.Runtime [Runtime.java:4318] in args --from-launcher --log-level info -s webgui WebGui intro Intro python Python 
10:20:45.336 [main] INFO c.m.s.Runtime [Runtime.java:4319] {"DEFAULT_CONNECT":"http://localhost:8888","autoUpdate":false,"daemon":false,"fromLauncher":false,"help":false,"logLevel":"info","logFile":"myrobotlab.log","services":["webgui","WebGui","intro","Intro","python","Python"],"virtual":false,"class":"org.myrobotlab.framework.CmdOptions"}
10:20:45.339 [main] INFO c.m.s.Runtime [Runtime.java:4321] 
   _____         __________      ___.           __  .____          ___.    
  /     \ ___.__.\______   \ ____\_ |__   _____/  |_|    |   _____ \_ |__  
 /  \ /  <   |  | |       _//  _ \| __ \ /  _ \   __\    |   \__  \ | __ \ 
/    Y    \___  | |    |   (  <_> ) \_\ (  <_> )  | |    |___ / __ \| \_\ \
\____|__  / ____| |____|_  /\____/|___  /\____/|__| |_______ (____  /___  /
        \/\/             \/           \/                    \/    \/    \/ 
            resistance is futile, we have cookies and robots ...
10:20:45.350 [main] INFO c.m.s.Runtime [Runtime.java:546] services [webgui, WebGui, intro, Intro, python, Python]
10:20:45.351 [main] INFO c.m.s.Runtime [Runtime.java:554] attempting to invoke : webgui of type WebGui
10:20:45.353 [main] INFO c.m.s.Runtime [Runtime.java:635] Runtime.createService runtime
10:20:45.387 [main] INFO c.m.f.MethodCache [MethodCache.java:154] caching Runtime's 431 methods and 232 declared methods
10:20:45.485 [main] INFO c.m.f.MethodCache [MethodCache.java:200] cached Runtime 430 methods with 408 ordinal signatures in 124 ms
10:20:45.650 [main] INFO c.m.f.Registration [Registration.java:49] creating registration for runtime@abrasive-zeroid - org.myrobotlab.service.Runtime
10:20:45.726 [main] INFO c.m.s.Runtime [Runtime.java:1707] runtime@abrasive-zeroid registering at abrasive-zeroid of type org.myrobotlab.service.Runtime
10:20:46.144 [main] INFO c.m.s.m.JoystickMeta [JoystickMeta.java:23] Joystick.getMetaData x86.64.windows  isArm() false
10:20:46.145 [main] INFO c.m.s.m.JoystickMeta [JoystickMeta.java:28] loading non-arm binaries
10:20:46.576 [main] INFO c.m.f.r.ServiceData [ServiceData.java:186] Returning serviceData.json from data\.myrobotlab\serviceData.json
10:20:46.582 [main] INFO c.m.f.r.Repo [Repo.java:94] create repo libraries
10:20:46.599 [main] INFO c.m.f.r.Repo [Repo.java:475] loading libraries\repo.json
10:20:46.656 [main] INFO c.m.f.r.Repo [Repo.java:502] loaded libraries\repo.json
10:20:46.657 [main] INFO c.m.i.FileIO [FileIO.java:1065] looking for Resource framework/ivy.xml.template
10:20:46.665 [main] INFO c.m.i.FileIO [FileIO.java:1065] looking for Resource framework/ivysettings.xml.template
10:20:46.671 [main] INFO c.m.f.Service [Service.java:2444] runtime new locale is tr-TR
10:20:46.828 [main] INFO c.m.s.Runtime [Runtime.java:2599] ============== args begin ==============
10:20:46.843 [main] INFO c.m.s.Runtime [Runtime.java:2609] jvmArgs [-Djava.library.path=libraries/native, -Djna.library.path=libraries/native, -Dfile.encoding=UTF-8]
10:20:46.846 [main] INFO c.m.s.Runtime [Runtime.java:2611] file.encoding UTF-8
10:20:46.851 [main] INFO c.m.s.Runtime [Runtime.java:2612] args [--from-launcher, --log-level, info, -s, webgui, WebGui, intro, Intro, python, Python]
10:20:46.853 [main] INFO c.m.s.Runtime [Runtime.java:2614] ============== args end ==============
10:20:46.858 [main] INFO c.m.s.Runtime [Runtime.java:2616] ============== env begin ==============
10:20:46.861 [main] INFO c.m.s.Runtime [Runtime.java:2622] PATH not defined
10:20:46.867 [main] INFO c.m.s.Runtime [Runtime.java:2627] JAVA_HOME not defined
10:20:46.873 [main] INFO c.m.s.Runtime [Runtime.java:2634] PROCESSOR_ARCHITECTURE=AMD64
10:20:46.876 [main] INFO c.m.s.Runtime [Runtime.java:2641] PROCESSOR_ARCHITEW6432 not defined
10:20:46.879 [main] INFO c.m.s.Runtime [Runtime.java:2643] ============== env end ==============
10:20:46.881 [main] INFO c.m.s.Runtime [Runtime.java:2645] ============== platform ==============
10:20:46.887 [main] INFO c.m.s.Runtime [Runtime.java:2647] 2023/01/03 10:20:44 - GMT - 2023/01/03 07:20:44
10:20:46.894 [main] INFO c.m.s.Runtime [Runtime.java:2648] pid 2252
10:20:46.901 [main] INFO c.m.s.Runtime [Runtime.java:2649] hostname desktop-erjın6s
10:20:46.905 [main] INFO c.m.s.Runtime [Runtime.java:2650] ivy [runtime,x86.64.windows]
10:20:46.907 [main] INFO c.m.s.Runtime [Runtime.java:2651] version 1.1.1050 branch cb257854c665b172e588df2cf913c2a4eacb937e commit cb257854c665b172e588df2cf913c2a4eacb937e build null
10:20:46.911 [main] INFO c.m.s.Runtime [Runtime.java:2652] platform manifest {Archiver-Version=Plexus Archiver, Build-Host=null, Build-Jdk=11.0.17, Build-Time=202301030322, Build-User=root, Build-Version=1050, Built-By=root, Created-By=Apache Maven 3.8.6, GitBranch=cb257854c665b172e588df2cf913c2a4eacb937e, GitBuildTime=2023-01-02T19:22:50-0800, GitBuildUserEmail=grog@myrobotlab.org, GitBuildUserName=grog, GitBuildVersion=0.0.1-SNAPSHOT, GitClosestTagCommitCount=12886, GitClosestTagName=1.0.119, GitCommitId=cb257854c665b172e588df2cf913c2a4eacb937e, GitCommitIdAbbrev=cb25785, GitCommitIdDescribe=1.0.119-12886-gcb25785, GitCommitIdDescribeShort=1.0.119-12886, GitCommitIdFull=null, GitCommitTime=2023-01-02T18:38:03-0800, GitCommitUserEmail=grog@myrobotlab.org, GitCommitUserName=GroG, GitDirty=false, GitRemoteOriginUrl=https://github.com/MyRobotLab/myrobotlab.git, GitTags=1.1.1049, Implementation-Version=1.1.1050, Main-Class=org.myrobotlab.service.Runtime, Major-Version=1.1.1050, Manifest-Version=1.0}
10:20:46.917 [main] INFO c.m.s.Runtime [Runtime.java:2653] platform [x86.64.windows}]
10:20:46.918 [main] INFO c.m.s.Runtime [Runtime.java:2654] version [1.1.1050]
10:20:46.920 [main] INFO c.m.s.Runtime [Runtime.java:2655] root [/C:/MRL/myrobotlab-1.1.1050/myrobotlab.jar]
10:20:46.921 [main] INFO c.m.s.Runtime [Runtime.java:2656] cfg dir [data\.myrobotlab]
10:20:46.922 [main] INFO c.m.s.Runtime [Runtime.java:2657] sun.arch.data.model [64]
10:20:46.924 [main] INFO c.m.s.Runtime [Runtime.java:2659] ============== non-normalized ==============
10:20:46.925 [main] INFO c.m.s.Runtime [Runtime.java:2660] os.name [Windows 10] getOS [windows]
10:20:46.926 [main] INFO c.m.s.Runtime [Runtime.java:2661] os.arch [amd64] getArch [x86]
10:20:46.937 [main] INFO c.m.s.Runtime [Runtime.java:2662] os.version [10.0]
10:20:46.942 [main] INFO c.m.s.Runtime [Runtime.java:2664] java.vm.name [Java HotSpot(TM) 64-Bit Server VM]
10:20:46.944 [main] INFO c.m.s.Runtime [Runtime.java:2665] java.vm.vendor [Oracle Corporation]
10:20:46.946 [main] INFO c.m.s.Runtime [Runtime.java:2666] java.specification.version [11]
10:20:46.947 [main] INFO c.m.s.Runtime [Runtime.java:2676] file.encoding [UTF-8]
10:20:46.949 [main] INFO c.m.s.Runtime [Runtime.java:2677] Charset.defaultCharset() [UTF-8]
10:20:46.954 [main] INFO c.m.s.Runtime [Runtime.java:2678] user.language [tr]
10:20:46.955 [main] INFO c.m.s.Runtime [Runtime.java:2679] user.country [TR]
10:20:46.956 [main] INFO c.m.s.Runtime [Runtime.java:2680] user.variant []
10:20:46.957 [main] INFO c.m.s.Runtime [Runtime.java:2684] java.home [C:\Program Files\Java\jdk-11.0.17]
10:20:46.959 [main] INFO c.m.s.Runtime [Runtime.java:2686] java.library.path [libraries/native]
10:20:46.960 [main] INFO c.m.s.Runtime [Runtime.java:2687] user.dir [C:\MRL\myrobotlab-1.1.1050]
10:20:46.964 [main] INFO c.m.s.Runtime [Runtime.java:2689] user.home [C:\Users\user]
10:20:46.965 [main] INFO c.m.s.Runtime [Runtime.java:2690] total mem [136] Mb
10:20:46.967 [main] INFO c.m.s.Runtime [Runtime.java:2691] total free [55] Mb
10:20:46.969 [main] INFO c.m.s.Runtime [Runtime.java:2696] guessed os bitness [64]
10:20:46.969 [main] INFO c.m.s.Runtime [Runtime.java:2703] getting local repo
[... 192 lines cut ...]
10:20:53.546 [Thread-3] INFO c.m.s.Runtime [Runtime.java:2023] releasing service webgui
10:20:53.547 [Thread-3] INFO c.m.s.Runtime [Runtime.java:1807] releasing service webgui@abrasive-zeroid
10:20:53.549 [Thread-3] INFO c.m.s.Runtime [Runtime.java:1878] unregister webgui@abrasive-zeroid
10:20:53.552 [Thread-3] INFO c.m.s.Runtime [Runtime.java:1917] released webgui@abrasive-zeroid
10:20:53.550 [webgui] INFO c.m.f.Service [Service.java:1987] webgui info shutting down
[... cut ...]
10:20:53.594 [Thread-3] ERROR c.m.f.Service [Service.java:1946] runtime error java.lang.NullPointerException threw while releasing
10:20:53.595 [Thread-3] INFO c.m.s.Runtime [Runtime.java:1807] releasing service runtime@abrasive-zeroid
10:20:53.596 [Thread-3] INFO c.m.s.Runtime [Runtime.java:1878] unregister runtime@abrasive-zeroid
10:20:53.597 [Thread-3] INFO c.m.s.Runtime [Runtime.java:1917] released runtime@abrasive-zeroid
10:20:53.597 [runtime] INFO c.m.f.Service [Service.java:1987] runtime info shutting down

supertick, 2023-04-02:
sorry, been a while since i looked at issues ...
It's worky for others are you still having problems ?

supertick, 2023-04-02:
it may have been you didn't get any services to install

closed 2023-04-02 by supertick