MyRobotLab/myrobotlab #1250: AudioFile service play actual stream

supertick, 2023-02-01:
Currently the AudioFile service downloads any internet sound file and will cache and play it.
However, this will not work with audio streams.
An example http://direct.fipradio.fr/live/fip-midfi.mp3 from lecagnois
In this case, the read buffer of Http service should go directly to the audioFile in a new stream thread.

hairygael, 2023-02-01:
There is another way to stream radio, is to use the oob tags in the AIML files if required.

supertick, 2023-02-01:
how ?   ... is python streaming a real stream or is it downloading a file and playing ?

hairygael, 2023-02-01:
It's directly the browser that streams. The aiml use webgui service to startBrowser with the link of the radio and that's it. The only pb is that webgui service doesn't seem to have stopBrowser. I have put a gif on discord showing it.

hairygael, 2023-02-01:
`<category><pattern>START RADIO</pattern>
<template>Ok I start the radio.  
        <oob>
          <mrl>
            <service>webgui</service>
            <method>startBrowser</method>
            <param>http://direct.fipradio.fr/live/fip-midfi.mp3</param>
          </mrl>
        </oob></template>
</category>`

supertick, 2023-04-02:
handling all different streams from the web is non trivial .. perhaps a BDD ticket for VLC

closed 2023-04-02 by supertick