MyRobotLab/myrobotlab #1268: Simulator not working

frankstolp80, 2023-04-20:
when starting simulator get blank on version myrobotlab-1.1.1061 is working after version myrobotlab-1.1.1069 it not working i tryd the last version myrobotlab-1.1.1071 still blank

[image]
[image]

supertick, 2023-04-20:
It looks more like your missing InMoov2 models - and this occurred because of maven repo shifting.
You can try this, on windows delete the .ivy2 located go to C:\Users\{yourUser}\
Start myrobotlab in a fresh directory.

frankstolp80, 2023-04-20:
@supertick i deleted the .ivy2 directory but after restart still blank

supertick, 2023-04-20:
Looks worky for me .. :thinking: 
[image]

supertick, 2023-04-20:
any errors in the myrobotlab-install.log ?

frankstolp80, 2023-04-20:
this is the myrobotlab-install.log

22:41:05.350 [main] INFO c.m.f.Platform [Platform.java:269] git loading properties from jar /E:/mrl/myrobotlab-1.1.1071a/myrobotlab.jar
22:41:05.717 [main] INFO c.m.s.Runtime [Runtime.java:4453] in args --install --log-file myrobotlab-install.log 
22:41:06.171 [main] INFO c.m.s.Runtime [Runtime.java:4454] {"DEFAULT_CONNECT":"http://localhost:8888","autoUpdate":false,"config":null,"connect":null,"help":false,"id":null,"install":[],"invoke":null,"jvm":null,"logLevel":"info","logFile":"myrobotlab-install.log","memory":null,"services":[],"virtual":false,"class":"org.myrobotlab.framework.CmdOptions"}
22:41:06.173 [main] INFO c.m.s.Runtime [Runtime.java:4456] 
   _____         __________      ___.           __  .____          ___.    
  /     \ ___.__.\______   \ ____\_ |__   _____/  |_|    |   _____ \_ |__  
 /  \ /  <   |  | |       _//  _ \| __ \ /  _ \   __\    |   \__  \ | __ \ 
/    Y    \___  | |    |   (  <_> ) \_\ (  <_> )  | |    |___ / __ \| \_\ \
\____|__  / ____| |____|_  /\____/|___  /\____/|__| |_______ (____  /___  /
        \/\/             \/           \/                    \/    \/    \/ 
            resistance is futile, we have cookies and robots ...
22:41:06.195 [main] INFO c.m.s.Runtime [Runtime.java:624] Runtime.createService runtime
22:41:06.206 [main] INFO c.m.f.Service [Service.java:623] constructing runtime
22:41:06.237 [main] INFO c.m.f.MethodCache [MethodCache.java:155] caching Runtime's 432 methods and 233 declared methods
22:41:06.296 [main] INFO c.m.f.MethodCache [MethodCache.java:201] cached Runtime 431 methods with 409 ordinal signatures in 67 ms
22:41:06.400 [main] INFO c.m.f.Registration [Registration.java:49] creating registration for runtime@irritating-galaxina - org.myrobotlab.service.Runtime
22:41:06.530 [main] INFO c.m.s.Runtime [Runtime.java:1705] runtime@irritating-galaxina registering at irritating-galaxina of type org.myrobotlab.service.Runtime
22:41:06.536 [main] INFO c.m.f.r.ServiceData [ServiceData.java:82] ================ generating serviceData.json begin ================
22:41:06.539 [main] INFO c.m.i.FileIO [FileIO.java:559] findPackageContents root [/E:/mrl/myrobotlab-1.1.1071a/myrobotlab.jar], src [org/myrobotlab/service], recurse [false], include [[.*\.class]], exclude [[]]
22:41:06.603 [main] INFO c.m.i.FileIO [FileIO.java:491] found 165 service files in /E:/mrl/myrobotlab-1.1.1071a/myrobotlab.jar
22:41:06.607 [main] INFO c.m.i.FileIO [FileIO.java:503] found 165 service classes
22:41:06.609 [main] INFO c.m.f.r.ServiceData [ServiceData.java:90] found 165 services
22:41:06.709 [main] INFO c.m.s.m.JoystickMeta [JoystickMeta.java:23] Joystick.getMetaData x86.64.windows  isArm() false
22:41:06.713 [main] INFO c.m.s.m.JoystickMeta [JoystickMeta.java:28] loading non-arm binaries
22:41:06.725 [main] INFO c.m.f.r.ServiceData [ServiceData.java:128] ================ generating serviceData.json end ================
22:41:06.726 [main] INFO c.m.f.r.ServiceData [ServiceData.java:390] saving data\.myrobotlab\serviceData.json
22:41:07.105 [main] INFO c.m.f.r.ServiceData [ServiceData.java:195] saved generated serviceData.json to data\.myrobotlab\serviceData.json
22:41:07.109 [main] INFO c.m.f.r.Repo [Repo.java:95] create repo libraries
22:41:07.114 [main] INFO c.m.f.r.Repo [Repo.java:499] libraries\repo.json not found
22:41:07.116 [main] INFO c.m.f.r.Repo [Repo.java:506] loaded libraries\repo.json
22:41:07.117 [main] INFO c.m.i.FileIO [FileIO.java:1065] looking for Resource framework/ivy.xml.template
22:41:07.120 [main] INFO c.m.i.FileIO [FileIO.java:1065] looking for Resource framework/ivysettings.xml.template
22:41:07.123 [main] INFO c.m.f.Service [Service.java:2569] runtime new locale is nl-NL
22:41:07.295 [main] INFO c.m.s.Runtime [Runtime.java:2716] ============== args begin ==============
22:41:07.298 [main] INFO c.m.s.Runtime [Runtime.java:2726] jvmArgs [-Djava.library.path=libraries/native, -Djna.library.path=libraries/native, -Dfile.encoding=UTF-8]
22:41:07.300 [main] INFO c.m.s.Runtime [Runtime.java:2728] file.encoding UTF-8
22:41:07.301 [main] INFO c.m.s.Runtime [Runtime.java:2729] args [--install, --log-file, myrobotlab-install.log]
22:41:07.303 [main] INFO c.m.s.Runtime [Runtime.java:2731] ============== args end ==============
22:41:07.304 [main] INFO c.m.s.Runtime [Runtime.java:2733] ============== env begin ==============
22:41:07.305 [main] INFO c.m.s.Runtime [Runtime.java:2739] PATH not defined
22:41:07.306 [main] INFO c.m.s.Runtime [Runtime.java:2744] JAVA_HOME not defined
22:41:07.307 [main] INFO c.m.s.Runtime [Runtime.java:2751] PROCESSOR_ARCHITECTURE=AMD64
22:41:07.307 [main] INFO c.m.s.Runtime [Runtime.java:2758] PROCESSOR_ARCHITEW6432 not defined
22:41:07.308 [main] INFO c.m.s.Runtime [Runtime.java:2760] ============== env end ==============
22:41:07.309 [main] INFO c.m.s.Runtime [Runtime.java:2762] ============== platform ==============
22:41:07.316 [main] INFO c.m.s.Runtime [Runtime.java:2764] 2023/04/20 22:41:05 - GMT - 2023/04/20 20:41:05
22:41:07.317 [main] INFO c.m.s.Runtime [Runtime.java:2765] pid 5392
22:41:07.317 [main] INFO c.m.s.Runtime [Runtime.java:2766] hostname drive-in_-_muziek
22:41:07.321 [main] INFO c.m.s.Runtime [Runtime.java:2767] ivy [runtime,x86.64.windows]
22:41:07.322 [main] INFO c.m.s.Runtime [Runtime.java:2768] version 1.1.1071 branch develop commit 144ee48b9e6c037e811c8d55efba55fc4928ce83 build null
[... cut ...]
23:40:01.635 [main] INFO c.m.f.r.Repo [IvyWrapper.java:512] C:\Users\Frank Stolp\.ivy2\cache\org.ejml\dense64\jars\dense64-0.27.jar
23:40:01.635 [main] INFO c.m.f.r.Repo [IvyWrapper.java:512] C:\Users\Frank Stolp\.ivy2\cache\org.ejml\simple\jars\simple-0.27.jar
23:40:01.635 [main] INFO c.m.f.r.Repo [IvyWrapper.java:512] C:\Users\Frank Stolp\.ivy2\cache\org.ejml\denseC64\jars\denseC64-0.27.jar
23:40:01.636 [main] INFO c.m.f.r.Repo [IvyWrapper.java:512] C:\Users\Frank Stolp\.ivy2\cache\jfugue\jfugue\jars\jfugue-5.0.7.jar
23:40:01.636 [main] INFO c.m.f.r.Repo [IvyWrapper.java:512] C:\Users\Frank Stolp\.

AutonomicPerfectionist, 2023-04-20:
Can you check if you have an `InMoov2` directory in the folder above where you installed MRL?

frankstolp80, 2023-04-20:
it is in the above directory see screenhot
[image]

supertick, 2023-04-21:
screenshot was not successful

frankstolp80, 2023-04-21:
@supertick it is in the above directory see screenhot i copyd the imoov2 the mrl\Inmoov2 and it works now
my sinstall directory is mrl\myrobotlab-1.1.1071a...

[image]

AutonomicPerfectionist, 2023-04-21:
As I was afraid of, that InMoov2 directory is not actually supposed to be there. @supertick we've seen multiple instances of that directory appearing now, I think we should investigate why it's being created there

supertick, 2023-04-21:
yeah @frankstolp80 .. do you know how the InMoov2 directory got there ?
when myrobotlab starts and that directory exists at that location it thinks int in "develop mode" ... which is probably not your intention.

frankstolp80, 2023-04-21:
@supertick  after a fresh install of myrobotlab it it creates the directory or how can i turn off the develop mode when i delete the directory it and restart it creates directory again than emty see screenshot

[image]

supertick, 2023-04-21:
checking ....

frankstolp80, 2023-04-21:
i got this error also

[image]

supertick, 2023-04-21:
i can't seem to recreate the problem using 1.1.1071
I don't get InMoov2 appearing  .. :thinking: 
[image]

frankstolp80, 2023-04-21:
the opencv has also errors wheni try to add filters see log

[myrobotlab.log](https://github.com/MyRobotLab/myrobotlab/files/11297671/myrobotlab.log)

frankstolp80, 2023-04-28:
it works now

closed 2023-04-28 by frankstolp80