MyRobotLab/myrobotlab #1270: opencv not saving after adding filter error

frankstolp80, 2023-04-21:
when i try add a filter  i get errors

[image]

[image]

[myrobotlab.log](https://github.com/MyRobotLab/myrobotlab/files/11299436/myrobotlab.log)

supertick, 2023-04-21:
Hi @frankstolp80, 
thanks for testing.  Yeah valid bug, I can imagine many of the opencv filters configuration is not serializable.  It will take some time to correct, and it might get superseded by InMoov2 state management because, it might be more useful to use state transitions to change filters.  For example, if InMoov senses movement and a state transition switches to attempt to recognize, then face recognition filters would be loaded at that time, vs loaded from starting config .

frankstolp80, 2023-04-22:
@supertick  when starting i get this error see screenshot

[image]

AutonomicPerfectionist, 2023-06-05:
> @supertick when starting i get this error see screenshot
@frankstolp80 sorry we haven't responded in awhile, development has been very busy. That error is indeed a bug but where it occurs looks to me like your Arduino was not connected, so if you see that specific error again that's why. I will try to fix the bug when I can so you get a more useful warning message.

For your earlier issues regarding the filters and configs, it looks to me like we will need to do some more in-depth work on specifically the OpenCV config code to fix it. @supertick it looks like the config uses a map between strings (I assume names of filters) and the actual filter objects. Looking at the stacktrace in the above myrobotlab.log it appears the main issue is SnakeYaml can't find a no-args constructor of the OpenCV service while deserializing the OpenCVFilterFaceDetector filter. The only reference to the service itself I see in that class is in the superclass OpenCVFilter, the field `opencv`. It appears it's marked transient but it has a getter and a setter, of which only the setter is marked with `@Transient`. I don't know if marking the getter with that as well will fix it, I don't have the offending config file, but it could be worth it to check