MyRobotLab/myrobotlab #1326: 错误: 找不到或无法加载主类 org.myrobotlab.service.Runtime 原因: java.lang.ClassNotFoundException: org.myrobotlab.service.Runtime

mrz5802, 2023-08-05:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

supertick, 2023-08-05:
get a later version https://build.myrobotlab.org:8443/job/myrobotlab/job/develop/1201/artifact/target/myrobotlab.zip

mrz5802, 2023-08-05:
Thank you, it has been resolved

closed 2023-08-05 by supertick