MyRobotLab/myrobotlab #1429: No GUI for Nixie2 with Ubuntu 24.04 LTS

Gleis1234, 2024-09-02:
**Describe the bug**
Starting myrobotlab.sh no GUI opens.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to $HOME/[myrobotlab.log](https://github.com/user-attachments/files/16839251/myrobotlab.log)
2. start myrobotlab.sh

**Expected behavior**
Opening a GUI in browser with Nixie2 content

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu 24.04 LTS
 - Browser firefox, chrome
 - Version 129.0.2, 128.0.6613.113

**Additional context**
Add any other context about the problem here.

supertick, 2024-09-02:
I often run it on 22.04,
what does the myrobotlab.log say ?

Gleis1234, 2024-09-08:
The logfile gives me no idea what's going wrong. Logfile added. Maybe anyone will find something.
[myrobotlab.log](https://github.com/user-attachments/files/16921123/myrobotlab.log)

supertick, 2024-09-08:
I see this ..
20:58:35.010 [main] WARN c.m.n.BareBonesBrowserLaunch [BareBonesBrowserLaunch.java:37] graphical browser requested but currently no display
Are you starting in a ssh session ?

Gleis1234, 2024-09-08:
No, I start from the console on the desktop of a Kubuntu installation. The Plasma environment of Kubuntu runs under Xorg.

supertick, 2024-09-08:
Do you have a browser on the desktop ?
If so, have you tried manually going to http://localhost:8888 ?

Gleis1234, 2024-09-08:
No chance I purged the system.

Gleis1234, 2024-09-08:
Back in 22.04 everything works as expected.

closed 2024-09-08 by Gleis1234

supertick, 2024-09-08:
From the logs it looks like MRL started without a problem on your Plasma environment of Kubuntu.
MRL is a web server, so I believe you could use it in that environment.  The only problem I saw, is MRL did not know how to "automatically" spawn a browser to open the URL .. which could be done manually or remotely.
Just a FYI if you want to go back to using Plasma environment of Kubuntu.